olikraus/u8glib #352: Supportinformation multiple device types on spi inte

jbhome, 2015-07-23:


olikraus, 2015-07-23:
yes agree... a very old design mistake...

jbhome, 2015-07-24:
Can it be fixed?
just add spi.Trans at start of transfer function?
I'd rather not hack your code.
any chance of your adding this?

I also have a question about write fisrst/last
why not just integrate within a single write function?
is there a good place to put questions like this? 
Thanks for your patience with a newbie

olikraus, 2015-07-24:
> Can it be fixed?
Planned for U8glib 2.0. However i do not know when i will have time to make a larger rework of u8glib.
> why not just integrate within a single write function?
This would require much more RAM, which is not there in some devices.
However, another option which is planed for 2.0

> is there a good place to put questions like this? 
I usually discuss and answer questions on the Arduino Forum (Display section)