olikraus/u8glib #354: Random pixel on 128x32 SainSmart Display

jstrobel, 2015-07-26:
Hi,
i've connected a SainSmart 0.91" 128x32 Display to an Arduino Pro mini. I use the i2c Pins A4 / A5 and VCC and GND. When i upload some example source with:
U8GLIB_SSD1306_128X32 u8g(U8G_I2C_OPT_NO_ACK);

I got the result as seen on the attached image. When i am using the adafruit-Library i got some normal output. What i am doing wrong?

Thanks
Jens
[image]

olikraus, 2015-07-26:
What happens with the GraphicsText example? Do you see some pixels moving around?

jstrobel, 2015-07-26:
After using the GraphicsText-Example and readding my source everything works now.
Sorry, can't reproduce why it hasn't worked before.

closed 2015-07-26 by jstrobel