olikraus/u8glib #355: Get bitmap from screen data

raffahacks, 2015-07-28:
I don't think it would be very difficult to code, but it would allow MANY more features, like for example animations that require the screen bitmap (am I clear in my explanation?)

olikraus, 2015-07-28:
You simply can not get a bitmap from the current screen data. This is not possible with most of the LCDs.

raffahacks, 2015-07-29:
Hello Olikraus,
Thanks for your answer. Sorry if I wasn't clear. What I mean is not getting
the screen bitmap from the screen directly, but instead telling u8glib to
store it in a variable (like XBMs) in the RAM while writing it on the
screen. An example:

u8g.firstPage();
u8g.displayToBitmap(OutputVariableNameHere);

do {
  // something
} while(u8g.nextPage());

Cheers

>
> You simply can not get a bitmap from the current screen data. This is not
possible with most of the LCDs.

olikraus, 2015-07-30:
ok, you want something like a virtual display, which outputs to RAM instead of the actual display.

raffahacks, 2015-07-31:
Yes, something like that. I think many people would use it, but, as I am
not an expert C programmer, I don't know how easy it would be to code...