olikraus/u8glib #356: Fix font data

olikraus, 2015-08-01:
u8g_font_data.c:18746:const u8g_fntpgm_uint8_t u8g_font_fub20t[477] U8G_SECTION(".progmem.u8g_font_fub20t") = {
--> shound not be U8G_SECTION

jbhome, 2015-08-02:
So then what should it be

>
> u8g_font_data.c:18746:const u8g_fntpgm_uint8_t u8g_font_fub20t[477]
U8G_SECTION(".progmem.u8g_font_fub20t") = {
> --> shound not be U8G_SECTION

Jim

olikraus, 2015-08-02:
The font should not be there at all.

jbhome, 2015-08-02:
Off topic but....
What are the spi settings for st7920 spi ?

olikraus, 2015-08-02:
which settings?

jbhome, 2015-08-02:
All settings associated with spi configuration
I would like to use spi.transaction but having difficulty finding where and
what settings are used to initialize the device

olikraus, 2015-08-02:
the SPI settings for the ST7920 are mentioned in the datasheet for the ST7920

jbhome, 2015-08-02:
Sorry what are the arduino settings you use in u8g?

olikraus, 2015-08-03:
you could study u8g_com_arduino_st7920_hw_spi.c

I have a remark in that file:

      /* 20 Dez 2012: set CPOL and CPHA to 1 !!! */
      SPCR =  (1<<SPE) | (1<<MSTR)|(0<<SPR1)|(0<<SPR0)|(1<<CPOL)|(1<<CPHA);

jbhome, 2015-08-03:
Ok
So you don't use the arduino spi configuration functions
Sorry to be a pain
With the library supporting so many configuration finding the right place
is challenging specially when it's in c++ and I'm still learning
Thanks for your help AND your patience
Sincerely
Jim

olikraus, 2016-03-05:
No problem. I hope to support Arduino standard SPI in the future.

olikraus, 2016-03-05:
the original issue is fixed

closed 2016-03-05 by olikraus