olikraus/u8glib #358: Even-diameter circles cannot be drawn

steronydh, 2015-08-05:
drawCircle, drawDisc etc. draw circles with a diameter of `(2 * radius + 1)`. This means that you cannot draw a circle with diameter 6, 8, 10 etc.

Could it be made possible to specify circles by diameter instead of radius, so that diameters can be even?

olikraus, 2015-08-05:
It is a property of the midpoint algorithm, that the circle has a odd diameter. The implementation of a different algorithm could solve this, but i am not aware about a different algorithm which can do this.

So the simple answer is: No not possible with the currenct implementation.

steronydh, 2015-08-06:
OK, no problem. It was only an issue because I am using small circles where single-pixel differences matter, but I can get around it by breaking them down and drawing them in bits.

closed 2016-03-17 by steronydh