olikraus/u8glib #361: otf2bdf for windows

ganeshkumarsw, 2015-08-17:
Is there any tool like otf2bdf for wndows 64 bit OS, I tried the binary version of the same released for windows but facing failure during the output stage.

If you have any solution kindly let me know.

olikraus, 2015-08-17:
hmmm.. i can not say much here. I only use ubuntu as a build environment.

ganeshkumarsw, 2015-08-17:
It would be good to have that same working for windows environment, people like us can be benefited by this. Looking forward for extended support here!!

olikraus, 2015-08-17:
Hmm... I am not the maintainer of otf2bdf. Do you think it makes sense to complain about a not running tool for a specific windows environment in the bug tracker of an Arduino library?
I am confused.

ganeshkumarsw, 2015-08-19:
Sorry that was a request, no offence. Since the tool was available under u8glib repo, I believe it make sense to post here request for your support.

[image]

 I never complained about your work, if you read my prev pos, what I have asked is for your extended support, so kindly don't take it as a complaint. 

When you post this kind of project in public repo and people start using it, you will get comments and in my opinion they are all for improvements not "complaints" but if you consider that as complaint then I dont know what to say.

**I am extremely sorry for requesting your support.**

closed 2015-08-19 by ganeshkumarsw

olikraus, 2015-08-20:
You are right, my fault. I forgot that i placed otf2bdf in the repo. Comments and opinons are highly welcome, but for otf2bdf i really can not do anything.

Thanks anyway for the discussion.

ganeshkumarsw, 2015-08-21:
No problem, I will find some other way.