olikraus/u8glib #362: Weird SH1106 132x64 compatibility

tprochazka, 2015-08-23:
In the u8g_dev_ssd1306_128x64.c are some comments about compatibility with SH1106.
But SH11064 has 132 pixel width. Why you don't support 132x64 directly, but only as centerer 128x64 scren?

Only neccesary change to support full screen size on this OLED is change 
#define WIDTH 128    --->     #define WIDTH 132

and

0x000,		/* set lower 4 bit of the col adr to 0  */       ---->  to 0x004

tprochazka, 2015-08-23:
Or not all SH11064 are 132x64 pixels?

olikraus, 2015-08-24:
Yes, indeed. All SH1106 displayes known to me are of size 128x64. This includes may far east displays, which are sold as SSD1306.

As soon as I have a SH1106 display with 132x64 in my lab i will add the constructor.

BTW: It is also surprising, that all existing 128x64 OLEDs have centered the display in the 132x64 RAM. I guess that there is finally only one source for the OLED...

tprochazka, 2015-08-30:
I have this one
http://www.aliexpress.com/item/1-3-inch-OLED-display-module-blue/1389516150.html
It is working only with changes that I mentioned in first post.

olikraus, 2015-08-30:
Looks like this display from e-bay:
http://www.ebay.com/itm/261465635352

i will try to order this. Maybe this has the same resolution. If this is true, i could check the modification.

tprochazka, 2015-09-01:
But on eBay is 128 x 64 in the description and it cost more.

olikraus, 2015-09-01:
Maybe description is wrong. In fact the controller name mentioned there is also wrong. It is not SSD1106. It must be SH1106 or SSD1306. 
But you are right. Picture looks identical, but maybe it is still 128x64.