olikraus/u8glib #365: SSD1331 device support

jej, 2015-10-09:
Hi all,

I see that very far away, people sell [65k color OLED display](http://www.ebay.fr/itm/0-95inch-RGB-OLED-A-SSD1331-96-64-Resolution-65K-Colorful-Horizontal-Pinheader-/251859806533) ([or here](http://www.aliexpress.com/item/color-recognition-sensor-TCS230-TCS3200-Color-sensor-Color-recognition-module-color-recognition-sensor/32351811186.html)) with SSD1331 controller. This device is [mentioned in this repository](https://github.com/olikraus/u8glib/search?utf8=%E2%9C%93&q=SSD1331) but [nothing says](https://github.com/olikraus/u8glib/wiki/device) it is compatible for now.

Do you think this display could work easily with u8glib? Or maybe it works already?

Thanks for your answers.

jej, 2015-10-09:
PS: the manufacturer datasheet says that the resolution is 128x64. So maybe it works out of the box with SSD1325 driver...

olikraus, 2015-10-09:
I once splitt the library and created ucglib for true color displays. ucglib supports SSD1331 and there is also a picture on the gallery page: https://github.com/olikraus/ucglib/wiki/gallery

jej, 2015-10-09:
How did I miss it!! Sorry I didn't know about this "coloured fork"...

Thanks a lot for this work. Cheers.

olikraus, 2015-10-10:
:-)