olikraus/u8glib #366: Warning on compile

ruggb, 2015-10-10:
I am using Adruino 1.6.5 and Marlin RC2 and the updated version of u8glib - 1.14
I get this warming when I compile
============
d:\Users\Bill\Documents\Arduino\libraries\U8glib\src\clib\u8g_rot.c:48:1: warning: initialization from incompatible pointer type [enabled by default]

 u8g_dev_t u8g_dev_rot = { u8g_dev_rot_dummy_fn, NULL, NULL };

 ^

d:\Users\Bill\Documents\Arduino\libraries\U8glib\src\clib\u8g_rot.c:48:1: warning: (near initialization for 'u8g_dev_rot.dev_fn') [enabled by default]

olikraus, 2015-10-11:
The warning can be ignored, but yes, it should be fixed once.

mickkn, 2016-01-26:
If you replace 
uint8_t u8g_dev_rot_dummy_fn(void *u8g, void *dev, uint8_t msg, void *arg)
{
	return 0;
}

with
#ifdef __XC8
uint8_t u8g_dev_rot_dummy_fn(void *u8g, void *dev, uint8_t msg, void *arg)
{
	return 0;
}
#else
uint8_t u8g_dev_rot_dummy_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, void *arg)
{
	return 0;
}
#endif

the warning disappears

olikraus, 2016-01-28:
actualy, i have not yet investigated this. But i also do see the warning.

Oliver

olikraus, 2016-03-05:
fixed

closed 2016-03-05 by olikraus

bolsoncerrado, 2017-06-13:
Just landed here after googleing 

"warning: initialization from incompatible pointer type
 u8g_dev_t u8g_dev_rot = { u8g_dev_rot_dummy_fn, NULL, NULL };"

I still have this issue on Arduino IDE 1.6.12 and u8glib u8glib_arduino_v1.18.1.zip

Is this normal?

Thanks!

olikraus, 2017-06-14:
Yes, it is fixed in U8g2.

bolsoncerrado, 2017-06-14:
Oh I thought I downloaded the most current version yesterday? I was wrong or is 2.0 not yet released?

jhkim1327, 2019-05-29:
> If you replace
> uint8_t u8g_dev_rot_dummy_fn(void *u8g, void *dev, uint8_t msg, void *arg)
> {
> return 0;
> }
>
> with
> #ifdef __XC8
> uint8_t u8g_dev_rot_dummy_fn(void *u8g, void *dev, uint8_t msg, void *arg) [...]

Many Thanks! 

it works! fixed!! Thanks a lot :)

kpdelucia, 2019-06-20:
I am not that adept at arduino library manipulation.  when I open the new Marlin 2.0 It only loads Marlin configuration .h and configuration adv.h   Is the edit above supposed to be performed by opening u8glib.h?  I cant open it from arduino.  I can choose add file then add the u8glib.h  but i cannot find the lines you say need correction.  Where do i find the "uint8_t u8g_dev_rot_dummy_fn(void *u8g, void *dev, uint8_t msg, void *arg)" to be replaced?

Thanks