olikraus/u8glib #371: 6pin i2c 0.96 oled works with adafruit code but not u8glib

s21rc, 2015-10-23:
Hi Olikraus, Greetings from Bangladesh and thanks for the nice small footprint driver for the oled. I have few 0.96 oled (cheap chines, pic attached) which has the following pin:
GND
VCC
SCL
SDA
RST
D/C

It works if I connect the RST and D/C pin to digital pin and using the below code. But I am unable to operate it using only 4 wire (except the RST, D/C pin). Can you please have a look and advise if I can static LOW or HIGH those pins so that it works with U8GLIB. I am willing to use it with a ATMEGA328P-PU chip and thus need a small driver. Thanks a lot. Fazlay

[image]
[image]

Working Code:

/*  OLED          arduino
     D0-----------10
     D1-----------9
     RST----------13
     DC-----------11
     VCC----------5V
     GND----------GND*/
int SCL_PIN=A5;//D0
int SDA_PIN=A4; //D1
int RST_PIN=4;//RST
int DC_PIN=3; //DC

 void LED_CLS(void);
 void LED_Set_Pos(unsigned char x,unsigned char y);//Set the coordinate
 void LED_WrDat(unsigned char data);   //Write Data
 void LED_P6x8Char(unsigned char x,unsigned char y,unsigned char ch);
 void LED_P6x8Str(unsigned char x,unsigned char y,unsigned char ch[]);
 void LED_P8x16Str(unsigned char x,unsigned char y,unsigned char ch[]);

 void LED_PrintBMP(unsigned char x0,unsigned char y0,unsigned char x1,unsigned char y1,unsigned char bmp[]); 
 void LED_Fill(unsigned char dat);
 void LED_PrintValueC(unsigned char x, unsigned char y,char data);
 void LED_PrintValueI(unsigned char x, unsigned char y, int data);
 void LED_PrintValueF(unsigned char x, unsigned char y, float data, unsigned char num);
 void LED_PrintEdge(void);
 void LED_Cursor(unsigned char cursor_column, unsigned char cursor_row);
 void LED_PrintLine(void);

const unsigned char F6x8[][6] =
{
  { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 },   // sp
  { 0x00, 0x00, 0x00, 0x2f, 0x00, 0x00 },   // !
  { 0x00, 0x00, 0x07, 0x00, 0x07, 0x00 },   // "
  { 0x00, 0x14, 0x7f, 0x14, 0x7f, 0x14 },   // #
  { 0x00, 0x24, 0x2a, 0x7f, 0x2a, 0x12 },   // $
  { 0x00, 0x62, 0x64, 0x08, 0x13, 0x23 },   // %
  { 0x00, 0x36, 0x49, 0x55, 0x22, 0x50 },   // &
  { 0x00, 0x00, 0x05, 0x03, 0x00, 0x00 },   // '
  { 0x00, 0x00, 0x1c, 0x22, 0x41, 0x00 },   // (
  { 0x00, 0x00, 0x41, 0x22, 0x1c, 0x00 },   // )
  { 0x00, 0x14, 0x08, 0x3E, 0x08, 0x14 },   // *
  { 0x00, 0x08, 0x08, 0x3E, 0x08, 0x08 },   // +
  { 0x00, 0x00, 0x00, 0xA0, 0x60, 0x00 },   // ,
  { 0x00, 0x08, 0x08, 0x08, 0x08, 0x08 },   // -
  { 0x00, 0x00, 0x60, 0x60, 0x00, 0x00 },   // .
  { 0x00, 0x20, 0x10, 0x08, 0x04, 0x02 },   // /
  { 0x00, 0x3E, 0x51, 0x49, 0x45, 0x3E },   // 0
  { 0x00, 0x00, 0x42, 0x7F, 0x40, 0x00 },   // 1
  { 0x00, 0x42, 0x61, 0x51, 0x49, 0x46 },   // 2
  { 0x00, 0x21, 0x41, 0x45, 0x4B, 0x31 },   // 3
  { 0x00, 0x18, 0x14, 0x12, 0x7F, 0x10 },   // 4
  { 0x00, 0x27, 0x45, 0x45, 0x45, 0x39 },   // 5
  { 0x00, 0x3C, 0x4A, 0x49, 0x49, 0x30 },   // 6
  { 0x00, 0x01, 0x71, 0x09, 0x05, 0x03 },   // 7
  { 0x00, 0x36, 0x49, 0x49, 0x49, 0x36 },   // 8
[... 675 lines cut ...]
 {  
  if(cursor_row != 0) 
  {
    if(cursor_column == 1) LED_Set_Pos(0, cursor_row + 2);
    else  LED_Set_Pos(80 + (cursor_column - 2)*6, cursor_row + 2);
    LED_WrDat(0xFF);
    LED_WrDat(0xFF);
    LED_WrDat(0xFF);
    LED_WrDat(0xFF);
    LED_WrDat(0xFF);
    LED_WrDat(0xFF);           
  }
 }
 void setup()
 {
   LEDPIN_Init();
   LED_Init();
 }
 void loop()
 {
  //  LED_P8x16Str(20,0,"Hello");
   // LED_P8x16Str(40,2,"World");
   // LED_P8x16Str(20,4,"Test");
   LED_P6x8Str(20,0,"Hello World");
 }

olikraus, 2015-10-24:
Did you try one of these constructors?
https://github.com/olikraus/u8glib/wiki/device#ssd1306-128x64
note that a0 equals d/c

Oliver

s21rc, 2015-10-24:
Hello,
Thanks for replying. I have tried all the three constructor for 1306 and also 1106. Now tested with : Analog 0 to D/C, 
Analog 0 to D/C, RST = GND
Analog 0 to D/C, RST = +5V

Nothing works. Please advise. Thanks a lot in advance.
Fazlay

s21rc, 2015-10-24:
BTW, it is a blue OLED with upper one row yellow.

olikraus, 2015-10-25:
Hi
I am confused. Your pictures show an SPI OLED, but you tested I2C? You should use the SPI constructors.

Oliver

raiman264, 2017-11-02:
I have the same display and digging around I fund that, regardless the pin names in the board, this is the real pinout that works with SPI libraries

`
  vcc gnd clk mosi rst dc
   |   |   |   |    |   |
   -----------------------
   |                     |
   |   My cheap OLED     |
   |                     |
   -----------------------
  
 `