olikraus/u8glib #377: 128x64 ST7920 SPI Atmega328p pixel problem

electricar, 2015-12-15:
Hello,

I'm trying to implement the u8glib in my project on an Atmega328p.
The problem is that I got messed up pixels in the first quarter of the lcd using the Atmega as an "normal" AVR without Arduino and it's bootloader. I don't have this problem when programming in the Arduino IDE with the bootloader.
Also in SW SPI mode as "normal" AVR, I don't have this pixel problem.
I already described my problem here, but nobody knew an answer:
https://www.mikrocontroller.net/topic/384277?goto=4385639#4385639

I hope that somebody here can help me.
If you need further information please let me know, as I'm really looking forward to get this working.

Thank you very much in advance!

Greets electricar

[image]

olikraus, 2015-12-16:
Hallo
Ausnahmsweise mal auf deutsch. Naja, ich hab das schon gelesen auf www.mikrocontroller.net. Geantwortet hab ich nicht, weil das meist wenig hilfreich war. Ob es an der U8glib liegt, kann ich freilich nicht ganz ausschließen, aber bei hunderten von usern (mich eingeschlossen) und vielen vielen anwendern von 3D-druckern läuft das ST7920 display ohne probleme.

Auf der anderen Seite, sind ST7920 basierte displays am häufigsten von dererlei problemen betroffen. Mir scheint der ST7920 controller hier besonders instabil zu sein.

Wie auch immer, manchmal hat es geholfen die Leitungslängen zu verkürzen.

Grüße,
Oliver

electricar, 2015-12-17:
Hallo Oliver,

vielen Dank für deine Antwort. Die Leitungslänge bekomme ich leider nicht kürzer und in der späteren Endapplikation wird diese auch wie jetzt ca. 10-15cm betragen. Also wenn das wirklich ein Problem ist, dann ist das LCD schlicht schlecht.
Auf der anderen Seite habe ich die gleichen Probleme mit einem zweiten Display (identisches Modell).
Komischerweise funktioniert die HW SPI Ansteuerung über Arduino ohne Probleme (das Board und Aufbau sind gleich, da ich den Bootloader lösche und ein normales hex-File auf den Atmega328p flashe).
SW SPI funktioniert ja auch, ist mir jedoch zu langsam in der Bildupdaterate.
Hast du evtl. noch eine andere Idee, wie ich die HW SPI ohne Arduino zum Laufen bekomme?
Vielen Dank im Voraus für deine Hilfe.
Gruß electricar

olikraus, 2015-12-17:
Hi
Sofern du den HW SPI code für den ATMega aus der u8glib benutzt, könnte es natürlich sein, dass da ein Fehler drin ist. Der Arduino HW SPI code für den AVR wurde in den letzten Jahren oft überarbeitet und perfektioniert. Um den "nicht-Arduino" Code hat sich keiner gekümmert. Vielleicht ist da ja irgendwas falsch. Es stimmt übrigens, das ST7920 schlecht ist... das sehe ich auch so.

Letztlich müsste man mal "u8g_com_atmega_st7920_hw_spi.c" und "u8g_com_arduino_st7920_hw_spi.c" bezüglich des HW setups vergleichen. Auf den ersten Blick von mir sieht es aber ganz gut aus.

Du könntest natürlich auch mal anfangen gezielt delays in den u8glib code einfügen.
Vieleicht hilft das ja. Konkret also in dieser routine: u8g_com_arduino_st7920_write_byte_hw_spi_seq

Oliver

electricar, 2015-12-17:
Hallo Oliver,

obwohl SW nicht gerade mein Steckenpferd ist, kann ich es zumindest mal versuchen...
Wobei ich fragen wollte, ob das nicht die falsche Routine ist? Arduino geht bei mir ja. Die Atmega HW SPI hat Probleme. Ist die entsprechende Routine folgende?:
u8g_com_atmega_st7920_write_byte_hw_spi

Viele Grüße

olikraus, 2015-12-18:
ach so ja, vermutlich...

RobertFach, 2018-01-09:
Hallo,
ich habe ähnliche Probleme mit dem Display. Ich betreibe es auch auf Hardware SPI und mit einen weiteren Slave. Interessanterweise streut der Slave auf das Display ein. Es scheint also entweder etwas mit den CS Signalen nicht korrekt zu sein, oder der Display Controller selbst verhält sich nicht korrekt auf den Bus...

Warum ich dies schreibe? Weil vermutlich hast du dasselbe Problem, soweit ich weiß schreibst du ja das Image über SPI (in diesem Moment hast du 2 Slaves auf den Bus). Gehst du jetzt über den Arduino Bootloader, dann kommt das Image über die Verbindung zum Bootloader. Auch war beim SW SPI der Bus wahrscheinlich nicht geshared. Es gibt auch ein Posting mit ähnlichen Beobachtungen von einem anderen Display Nutzer...

https://github.com/olikraus/u8g2/issues/367
https://www.eevblog.com/forum/projects/how-to-properly-connect-lcd-128x64-via-spi/
http://forum.arduino.cc/index.php?topic=181039.0

Cheers

Polle0, 2018-03-17:
Hello...
https://github.com/olikraus/u8g2/issues/367#event-1272813635