olikraus/u8glib #379: U8Glib & strcat issue

KiKiZ59, 2016-01-10:
Hello,
I'm dealing with a weird issue in my project using U8Glib and ST7920 screen.

In the setup function of the arduino sketch, i use strcat to complete and send a message thanks to 433MHz transmitter. I want the message to have a size of 70 chars so I complete it with some ';'.

The problem is : in the loop function, the text display on the screen is not the one i want to. Instead of my text, some words are replaced by ';;;;;;;;;'. That is due to the strcat function in the setup because when i change ';' by '!', the display in the loop change as well. This is weird because in the setup, i display other things after the strcat and there is no problem.

Here is a part of the code :
SETUP CODE
for(i=deb;i<fin;i++){   
      for(j=fin-deb;j<20;j++){
        strcat(message,";");strcat(message,";");strcat(message,";"); //IF i replace ';' by '!' => display on //the screen during the loop
      } 
      //Envoi du message Reçu
      vw_send((uint8_t *)message, strlen(message));
      vw_wait_tx();
      *message='\0';
      delay(200);
  }

C Function called in LOOP CODE
   int w=0,x=0,y=0;
   short decal=0;

   //Formation de la grille

   u8g.drawVLine(0,0,64); //x, y, longueur droite verticale
   u8g.drawHLine(0,0,128); //x, y, longueur droite horizontale
   u8g.drawVLine(63,0,64);
   u8g.drawHLine(0,31,128);
   u8g.drawVLine(127,0,64);
   u8g.drawHLine(0,63,128);

   //Affichage du texte

   // point de référence en haut à gauche pour drawStr et en bas à gauche pour print
   u8g.drawBox(0,0,128,11);
   u8g.drawBox(0,31,128,11);

   u8g.setColorIndex(0);
   u8g.setFont(u8g_font_4x6);//Choix police / Gras taille 7
   w = u8g.getStrWidth("TEMPS");
   x=floor((63-w)/2); y=3;//pour centrer
   u8g.drawStr(x,y,"TEMPS");//REPLACED BY ';;;;;;;;;;'   WHY??????????????????

Other C functions are called in the loop but only that one has that issue.
I don't know why. Is anyone able to help me?
Here are two pictures, the first one with strcat in the setup (transmitter) and the second without strcat (receiver).
[image]
[image]

KiKiZ59, 2016-01-10:
Hi,
I just fixed the problem. I forgot to allocate char* message ...
I have another question => Is it a problem if i don't delete "message" (delete [] message) in the sketch?
I can't do it because i use "message" in the loop (i can send message anytime).
Is it bad for the arduino memory?
KiKiZ

olikraus, 2016-01-10:
Hi

instead of new/delete it is usually better to allocate a fixed amount of
memory with

char message[128];

or so...

Oliver

KiKiZ59, 2016-01-10:
Hi,
Thanks for your reply.
I need to char* because i have to send message thanks to :
        vw_send((uint8_t *)message, 25);
        vw_wait_tx();
Is it bad for the arduino memory? Or the memory will be cleaned at the reset of the arduino?
KiKiZ

olikraus, 2016-01-10:
Hi

From using this, char message[100]; is the same as char *message.
I mean werever char * is expected you can provide also char [].

Oliver