olikraus/u8glib #380: Nodemcu and ssd1327 i2c

Sambutcher, 2016-01-12:
Hello,

I tried to made a seeedstudio Grove 96' oled screen (based on ssd1327) work on nodemcu v1.0.

I compiled a firmware with u8glib ssd1327_96x96_gr_i2c. Connectivity parameters are: SDA on pin 2, SDL on PIN 1 (due to Grove connection), and sla=0x3c.

Alter initialIsing i2c, only the declaration (disp=u8g.u8glib ssd1327_96x96_gr_i2c(0x3c)) made the screen react (Black and white dots), but I'm unable to write or change anything then with u8g subfunctions. Strangely, when I send i2c commands directly (with i2c.write ...), it seems to work (I was able to turn the screen off for exemple).

To complete, I tried before with the Seeedstudio library for Arduino and it works well (but for Arduino code and I want to use LUA!)

Any ideas about this issue?

Thank you for help,

Samuel

devsaurus, 2016-01-12:
@Sambutcher I'd propose to investigate this issue first in the context of NodeMCU since it might be more an integration topic than a native u8glib issue.
If you agree then let's continue at devsaurus/nodemcu-firmware#8 to avoid unnecessary traffic here.

devsaurus, 2016-03-05:
This issue can be closed.

olikraus, 2016-03-06:
ok

closed 2016-03-06 by olikraus