olikraus/u8glib #381: Mirroring screen by modifying u8g_rot.c

Salanova, 2016-01-12:
Hi olikraus,

I've seen some issues in which you said that rotation function could be modified in order to mirror the screen. (First of all, I'm using SD1306 128x64)

So I tried to modify 180 rotation by changing this:

        
        /* transform the reference point */
        //y = u8g_GetHeightLL(u8g, rotation_chain);
        y = u8g->height;
        y -= bbx->y; 
        y--;
        
        //x = u8g_GetWidthLL(u8g, rotation_chain);
        x = u8g->width;
        x -= bbx->x; 
        x--;
        
        /* adjust point to be the uppler left corner again */
        y -= bbx->h;
        y++;
        
        x -= bbx->w;
        x++;
                
        /* store x,y */
        bbx->x = x;
        bbx->y = y;
      }

To this

        
        /* transform the reference point */
       
        y = bbx->y; 
    
        
        //x = u8g_GetWidthLL(u8g, rotation_chain);
        x = u8g->width;
        x -= bbx->x; 
        x--;
        
        /* adjust point to be the uppler left corner again */
      
        
        x -= bbx->w;
        x++;
                
        /* store x,y */
        bbx->x = x;
        bbx->y = y;
      }

But it doesn't work... Any idea?

Thank you!!

olikraus, 2016-01-12:
Hi

I think more than just this little piece of code has to be changed.

Maybe it is easier to change the scan direction of the oled itself. I mean
change the value in line 96 from 0x0c8 to 0x0c0 in file
u8g_dev_ssd1306_128x64.c

Oliver

Salanova, 2016-01-13:
[image]

AWESOME!!!

Thank you so much!!!

olikraus, 2016-01-17:
cool

closed 2016-01-17 by olikraus