olikraus/u8glib #382: Hardware SPI D/C controlled by analog pin gives garbled screen (ok with digital pin)

MoonCactus, 2016-01-13:
This is on an Arduino Nano V3 with hardware SPI, CS on digital 5 and D/C on analog A6.
The display is a 128x64 Heltec 6 pin 128x64 display (VCC  GND  CLK  MOSI  CS  D/C), all white, bought on ebay. Works in 5V.

Tested with HelloWorld example. Got random pixels on the display, and dynamic/chaotic display with bits of readable text when I touched the A6 pin with a finger. To me, it looked like the analog pin was floating (checked with a scope: lots of noise, and display refreshed chaotically when I leave a finger on the pin).
I tried to add a 5.5K pullup with no better result though.

U8GLIB_SSD1306_128X64 u8g(5, A6);  // NOK

As soon as I used a digital pin to control D/C everything ran smoothly:
U8GLIB_SSD1306_128X64 u8g(5, 6); // OK

Great lib! Thanks

olikraus, 2016-01-17:
Hmm, i can not say much here. Looks more like a hw problem.

MoonCactus, 2016-03-06:
Sorry for getting back to it only now. I was not able to reproduce it on another Arduino nano.
I still got the issue with the original board: this analog pin is probably be faulty instead, so better consider this a false alarm, sorry for the report :/

closed 2016-03-06 by MoonCactus

flix9, 2022-09-05:
I am looking everywhere for the right documentation/code for this display that someone gave me, It seems to be the same one as you have, but I cannot find any mention of pins that read VCC GND CLK MOSI CS D/C in the documentation

Also in the helloworld example I cannot find any mention of analog pins...

olikraus, 2022-09-05:
U8glib is closed, please post your question to u8g2. Also add a picture of your display.