olikraus/u8glib #385: 1.3" OLED skipping lines

derekr27, 2016-01-29:
Hi there - bought an I2C SH1106 1.3" OLED from Creatron Inc in Canada (only OLED they sell). It worked fine so I removed it for a while. When I next hooked it up it seems to be skipping every other line. Am using Arduino IDE 1.6.7.

A 0.96" OLED substituted in it's place works perfectly, no skipped lines.

Here's a link: https://www.creatroninc.com/product/128x64-i2c-oled-display/ 

Anything I should check?

flav1972, 2016-02-01:
Hi,

I have the same issue with a SSD 0.96: it 128x64 using hardware SPI. Another similar module from another seller is working perfectly.

Initialised with:
U8GLIB_SSD1306_128X64 u8g(8, 9, 10);
Here is the image given:
[image]

A little search gave me this link: http://forum.espruino.com/conversations/269330/

I cannont know if it is a lib issue or a module issue.

derekr27, 2016-02-04:
Thanks for that, flav1972. I'm in the same boat as you - I checked the display carefully and there is damage in the lower left so I suspect I, too, have severed traces.

olikraus, 2016-02-07:
hmm... so i guess this not a lib problem and the issue can be closed.

flav1972, 2016-02-07:
for me you can close

derekr27, 2016-02-08:
I think so as it did work before. I’ve ordered another so will know for sure when it arrives.

 

Thanks

Derek

cunidev, 2016-02-13:
Try to replace the sh1106 constructor with some ssd1306 constructor. I experienced 2 pixels shifting on the x axis, and replacing sh1106 woth ssd1306 worked for me

Ariffffff, 2017-02-12:
I have same issue. I have checked library for height and used around 4-5 other libraries other than adafruit. should I believe my cheap module to be faulty ? or coding problem, If i remember correctly, it worked when i bought it few months ago attached are adafruit and U8 library demos
[image]

[image]

Ariffffff, 2017-02-12:
@derekr27 

Any updates on oled

olikraus, 2017-02-12:
Can you try this with u8g2?
Better also create an issue there.

Ariffffff, 2017-02-12:
thanks, sure thing, rightaway

zsoltmazlo, 2018-03-15:
This problem usually occurs if the display is broken, especially the bottom part (where there is no screen but glass). If anyone have a solution, do not hesitate to share with us!

flav1972, 2018-08-17:
all the modules in the pictures are broken.
=> Hardware issue (the connections are transparent in the glass).

jians5, 2019-01-24:
Guys, it's not broken. I just picked up my first OLED and it was skipping lines. Tutorials were telling me the library needs to be edited, but that's old pre-update information. Here's the real problem...

The Adafruit example sketch for 128x64 is broken. Use the 128x32 example sketch and change:

`define SCREEN_HEIGHT 64` to `#define SCREEN_HEIGHT 64` and it works fine.

I really hope I'm not the first person to figure this out and people have been throwing away perfectly good OLEDs.

Ariffffff, 2019-01-24:
Mine was broken. And even new failed shortly. Horrible option to use in
audio projects due to internal switching creating lot of noise. 2 bad
experiences. Ill srick with 16x2 and nokia for now

jians5, 2019-01-24:
OK - Just wanted to leave it here for future visitors looking for solutions.

BTW, Were you aware that you can put a 10k pullup resistor on the SCL and SDA ports to stabilize them? No more screen locking!

CriusDigitalStudio, 2022-01-11:
Indeed its broken I also fioud out the cause of mine skipping lines and it happened without me noticing after trying to fit the screen in a 3d printed case that it stuck into and i pulled it with force.

olikraus, 2022-01-11:
It would be cool to know how "u8g2" behaves here. The skipped line problem has been addressed in u8g2 with a special constructor.