olikraus/u8glib #386: Teensy 3.2 + SSD1306 + U8GLIB = Not working

FHdB, 2016-01-30:
I have a code that displays perfectly on a 128x64 SSD OLED when using Arduino Uno or Due over I2C.

I want to use it with a Teensy 3.2 with the same code and I haven't succeeded so far, other U8GLIB examples don't work either. There are no compiler errors. I have confirmed I2C to work and being correctly connected by using the i2c_scanner Arduino example (address: 0x3C)

I glanced into the u8g_com_i2c.c file and recognized that the teensy seems to take the "__AVR__" path and for that won't work because it can't use AVR commands. 

Is there any way to use the Teensy-specific "Wire.h" instead?

olikraus, 2016-02-07:
I have tested Teensy only once long time back. Unfortunately i can not do much here. Mostly because i do not know the Teensy good enough.

FHdB, 2016-02-07:
Mabye you have tried an AVR-Teensy? Those should work fine. Teensy 3.x rely on Freescale ARM Cortex-M4. Have I found the right file in "u8g_com_i2c.c" ? I would try to add Freescale-Teensy capability but I'm new to this. Is there any documentation / commentary on what exactly these functions do? I'm mostly unclear about what _exactly_ u8g_i2c_wait(uint8_t mask, uint8_t pos) does.

olikraus, 2016-02-07:
A wrapper is in general required, independent from the question, whether this is C++ or C.
Indeed, this is one of the biggest problems of u8glib, it is too deeply connected with AVR. I currently work to make these things better with the next version, but this will take a while.

FHdB, 2016-02-07:
As I said, I'm quite new to this. Why is a wrapper required generally?

Due to the number of Arduino compatible boards, it would make things easy to allow Wire.h to be used. At least as an alternate mode. I'd very much like to help implementing this.

olikraus, 2016-02-08:
A wrapper is required to allow I2C/TWI on none-Arduino boards. But i agree, it would be better, if U8glib supports Wire.h. Actually it is planed, but it will require some time, moreover it will require major rework on the library.

No1089, 2017-05-24:
This still seems to be a problem.
Just to confirm - because the Teensy is an ARM device, the "AVR only" u8Glib will not work with it?
Oliver - if you have never tried it, get a Teensy, they are wonderful to work with!

olikraus, 2017-05-24:
I rewrote u8glib to have support for all the none-AVR boards outside. U8g2 should indeed support Teensy and many more Arduino boards.

vasu15, 2018-04-04:
`
#include <u8g_i2c.h>
U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NO_ACK);  // Display which does not send AC
int frame=0;
const uint8_t frame1[] PROGMEM = {
0x00, 0x00, 0x00, 0x00, 0x0F, 0xFF, 0xFE, 0x00, 0x3F, 0xFF, 0xFF, 0x00, 0x3F, 0xFF, 0xFF, 0x80,
0x30, 0x00, 0x01, 0x80, 0x33, 0xFF, 0x81, 0xC0, 0x33, 0xFF, 0x81, 0xE0, 0x33, 0xFF, 0x81, 0xE0,
0x33, 0xFF, 0x81, 0xE0, 0x33, 0xFF, 0x81, 0xE0, 0x33, 0xFF, 0x81, 0xE0, 0x33, 0xFF, 0x81, 0xE0,
0x33, 0xFF, 0x81, 0xE0, 0x33, 0xFF, 0x81, 0xC0, 0x30, 0x00, 0x01, 0x80, 0x3F, 0xFF, 0xFF, 0x80,
0x3F, 0xFF, 0xFF, 0x00, 0x0F, 0xFF, 0xFE, 0x00, 0x00, 0x00, 0x00, 0x00,  
};

const uint8_t frame2[] PROGMEM = {
0x00, 0x00, 0x00, 0x00, 0x08, 0x00, 0x00, 0x0C, 0x00, 0x00, 0x0F, 0x00, 0x00, 0x09, 0x80, 0x00,
0x08, 0xE0, 0x00, 0x08, 0x38, 0x00, 0x08, 0x1C, 0x0C, 0x08, 0x0F, 0x06, 0x08, 0x1C, 0x03, 0x88,
0x70, 0x00, 0xC8, 0xE0, 0x00, 0x79, 0x80, 0x00, 0x1E, 0x00, 0x00, 0x1E, 0x00, 0x00, 0x79, 0x80,
0x00, 0xC8, 0xE0, 0x03, 0x88, 0x70, 0x06, 0x08, 0x1C, 0x0C, 0x08, 0x0F, 0x00, 0x08, 0x1C, 0x00,
0x08, 0x38, 0x00, 0x08, 0xE0, 0x00, 0x09, 0xC0, 0x00, 0x0F, 0x00, 0x00, 0x0C, 0x00, 0x00, 0x08,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 
};

const uint8_t frame3 [] = {
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x00, 0x00, 0x01, 0x80, 0x00, 0x00, 0x01, 0x80, 0x00,
0x00, 0x01, 0xC0, 0x00, 0x00, 0x01, 0xC0, 0x00, 0x00, 0x03, 0xC0, 0x00, 0x00, 0x03, 0xC0, 0x00,
0x00, 0x03, 0xC0, 0x00, 0x00, 0x03, 0xC0, 0x00, 0x00, 0x02, 0x40, 0x00, 0x00, 0x06, 0x40, 0x00,
0x00, 0x06, 0x60, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0x04, 0x60, 0x00,
0x00, 0x0C, 0x60, 0x00, 0x00, 0x0C, 0x60, 0x00, 0x07, 0xFC, 0x61, 0xFC, 0x07, 0xF8, 0x33, 0xFC,
0x00, 0x00, 0x33, 0x00, 0x00, 0x00, 0x33, 0x00, 0x00, 0x00, 0x33, 0x00, 0x00, 0x00, 0x33, 0x00,
0x00, 0x00, 0x36, 0x00, 0x00, 0x00, 0x36, 0x00, 0x00, 0x00, 0x3E, 0x00, 0x00, 0x00, 0x1E, 0x00,
0x00, 0x00, 0x1C, 0x00, 0x00, 0x00, 0x1C, 0x00, 0x00, 0x00, 0x1C, 0x00, 0x00, 0x00, 0x1C, 0x00,
0x00, 0x00, 0x18, 0x00, 0x00, 0x00, 0x18, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
};

const uint8_t frame4[] PROGMEM = {
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7F, 0xE0, 0x00, 0x00, 0x01, 0xFF, 0xF8, 0x00, 0x00, 0x07,
0xFF, 0xFE, 0x00, 0x00, 0x1F, 0xFF, 0xFF, 0x00, 0x00, 0x3F, 0xFF, 0xFF, 0x80, 0x00, 0x7F, 0xFF,
0xFF, 0xC0, 0x00, 0xFF, 0xFF, 0xFF, 0xF0, 0x01, 0xFF, 0xFF, 0xFF, 0xF0, 0x01, 0xFF, 0xFF, 0xFE,
0xF8, 0x03, 0xFF, 0xFF, 0xFC, 0x78, 0x03, 0xFF, 0xFF, 0xF8, 0x7C, 0x07, 0xFF, 0xFF, 0xF0, 0xFC,
0x07, 0xFF, 0xFF, 0xE1, 0xFC, 0x0F, 0xFF, 0xFF, 0xC3, 0xFE, 0x0F, 0xFF, 0xFF, 0x87, 0xFE, 0x0F,
0xFF, 0xFF, 0x0F, 0xFE, 0x0F, 0xF3, 0xFE, 0x1F, 0xFE, 0x0F, 0xE1, 0xFC, 0x3F, 0xFE, 0x0F, 0xE0,
0xF8, 0x7F, 0xFE, 0x0F, 0xF0, 0x70, 0xFF, 0xFE, 0x0F, 0xF8, 0x21, 0xFF, 0xFE, 0x0F, 0xFC, 0x03,
0xFF, 0xFE, 0x07, 0xFE, 0x07, 0xFF, 0xFC, 0x07, 0xFF, 0x0F, 0xFF, 0xFC, 0x07, 0xFF, 0x9F, 0xFF,
0xFC, 0x03, 0xFF, 0xFF, 0xFF, 0xF8, 0x01, 0xFF, 0xFF, 0xFF, 0xF8, 0x01, 0xFF, 0xFF, 0xFF, 0xF0,
0x00, 0xFF, 0xFF, 0xFF, 0xE0, 0x00, 0x7F, 0xFF, 0xFF, 0xC0, 0x00, 0x3F, 0xFF, 0xFF, 0x80, 0x00,
0x1F, 0xFF, 0xFF, 0x00, 0x00, 0x0F, 0xFF, 0xFE, 0x00, 0x00, 0x03, 0xFF, 0xF8, 0x00, 0x00, 0x00,
0x7F, 0xC0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
};
void setup(void) {
}

void loop(void) {
  u8g.firstPage();
 do{
    draw();
 }while(u8g.nextPage());
 frame++;
 if(frame == 13)
 frame=0;
 delay(1000);
}
void draw(){
 if(frame == 0 )
   {  u8g.drawBitmapP(0,0,4,20,frame1);
      u8g.drawBitmapP(103,0,3.5,28, frame2); 
      u8g.setFont(u8g_font_helvR24);
[... 108 lines cut ...]
      u8g.setColorIndex(1);
      u8g.drawStr(60,58,"2");
      u8g.setFont(u8g_font_profont22);
      u8g.drawStr(90,58,"sec");
   } 
   if(frame == 11 )
   {  u8g.setFont(u8g_font_profont22);
      u8g.setColorIndex(1);
      u8g.drawStr(10,20,"Analysing");
      u8g.drawBitmap(10,30,4,32, frame3);
      u8g.setFont(u8g_font_helvR24);
      u8g.setColorIndex(1);
      u8g.drawStr(60,58,"1");
      u8g.setFont(u8g_font_profont22);
      u8g.drawStr(90,58,"sec");
   } 
   if(frame == 12)
   {  u8g.drawBitmapP(45,0,5.375,43, frame4);
       u8g.setFont(u8g_font_profont22);
      u8g.setColorIndex(1);
      u8g.drawStr(45,58,"Done");
   } 
 }
`
This Code Works fine with UNO but not with teensy3.2 please help me to get out from this.

olikraus, 2018-04-04:
Unfortunately I can not help here...

vasu15, 2018-04-04:
Then where we can talk. Please help me out.
Tell me one thing how to use 4 pin oled with u8g2, because the methods you
have written are for 7 pin oled display.

olikraus, 2018-04-04:
> Then where we can talk. Please help me out.

As I wrote, I have no idea about Teensy. 
In general, if you have a OLED display, configured as I2C, then connect the same to the I2C Bus of your Teensy Board.