olikraus/u8glib #387: Support for LC7981 128x128

nixgadget, 2016-02-07:
Hi there, I own one of these LCDs https://www.crystalfontz.com/products/document/296/CFAG128128B-TMI-VZ_Data_Sheet_Preliminary_2012-04-17.pdf. 

I created the following to support the LCD to work with Arduino.
[u8g_dev_lc7981_128x128.txt](https://github.com/olikraus/u8glib/files/120673/u8g_dev_lc7981_128x128.txt)

However, when displaying i see some clipping (as shown in the screenshot)
[image]
. It looks like some of pixels have shifted. Can you please give me some tips on how to fix this ?

olikraus, 2016-02-07:
difficult, guess the memory mapping (RAM towards Display) is somehow odd.

nixgadget, 2016-02-08:
seems fine if i use 160x160. strange.