olikraus/u8glib #391: SCK pulses low in voltage

maukcc, 2016-02-23:
I have been using DOGM128 displays in our machines for some time. they are run by arduino MEGA in SW spi.
When using cables that are longer then 1meter we sometimes experience that the displays start to "scroll" , displays gibberish or stops working all together. 
After measuring the signals to the display we found that 3 signals (MOSI,CS and C0) are top 5V, but the SCK is top 1,2V at best (also with the "hello world" and other examples)
This is below specs for the dogm series display (and probably others as well) and a wonder that it works at all.
Is this feature? Or is this bug?

olikraus, 2016-03-05:
I do not think that this is a software problem of u8glib.

maukcc, 2016-03-07:
So is it programmed in the arduino SPI code?
Or is it a case of not/bad defined "digitalWrite(pin, HIGH);"
I tried the SCK on a normal output pin, on a PWM pin and on a analog pin and all 3 give the same result (with nothing connected to it). As the MOSI pin has the same 2.5MHz signal and does reach 5V top, the SCK pin should also be able to reach the 5V top signal. 
If the output signal could be programmed to a certain top value, the best would be to have it at 3V3 and not 5V

olikraus, 2016-03-28:
Hmmm the digitial pins can not be programmed to a specific voltage level. I guess there is some issue with the display.

Oliver

maukcc, 2016-03-29:
That is hard to understand as there is no display connected :)
I am measuring on open pins on the Arduino mega. (nothing is connected to the pins)
Could it be something with pullup and pulldown resistors that are active?
Also, is it possible to slow the SPI down from 2,5Mhz to 1Mhz (ish)

maukcc, 2016-03-31:
[image]
It is just the MEGA connected to 12V and GND and the hello world example loaded.
Nothing attached further, and the SCK pin only goes up to 1.2V
[image]

olikraus, 2016-04-04:
I probably can not say much here. Will a plain digitalWrite() work?

maukcc, 2016-04-05:
I would love to try the plain "digitalWrite()"  :)
The strange thing I am observing is that the MOSI signal(same Mhz) goes up to 5V and some spikes make it up to 5V and some do not. That is normal behaviour. 
But this SCK signal looks like it is deliberately chopped of at 1.2V.

I would also love to try out a lower SCK clock pulse(now 2.5Mhz or 400ns) to lets say 500Khz or 2000ns.
This so we can attach longer wires between controller and display

olikraus, 2016-04-05:
so, then, why don't you do so?

maukcc, 2016-04-06:
I would not know where to start. I am an electronics person and have minimal coding skills.

olikraus, 2016-04-10:
Well, probably beyond the scope of u8glib, but a script could look like this:

``` c
int pin = 10;   // related pin
void setup() {
  pinMode(pin, OUTPUT);
}
void loop() {
  // 100 kHz signal (10 micro seconds)
  digitalWrite(pin, 1)
  delayMicroseconds(1);   // high time is 1 micro seconds
  digitalWrite(pin, 0)
  delayMicroseconds(9);
}
```

HappyCodingRobot, 2016-04-18:
I think your 1.2V level could be a measurement problem.. If your SCK is 2.5MHz you should measure at a time base of e.g. 50ns. A sample rate of 100kS/s (10us) is way to low. If the sample and hold stage also is slow, this could lead to the 'analog' readings of the digital port.
Please check it much shorter time base or lower sck  (Nyquist ..)