olikraus/u8glib #392: Array of U8G_PROGMEM

igi79, 2016-02-25:
I've tried to rewrite menu demo using drawStrP method on Arduino:

const u8g_pgm_uint8_t string_0[] U8G_PROGMEM  = "First Line";
const u8g_pgm_uint8_t string_1[] U8G_PROGMEM  = "Second Item";
const u8g_pgm_uint8_t string_2[] U8G_PROGMEM  = "3333333";
const u8g_pgm_uint8_t string_3[] U8G_PROGMEM  = "abcdefg";
const u8g_pgm_uint8_t* const menu_strings[] U8G_PROGMEM = {string_0, string_1, string_2, string_3};

void drawMenu(void) {
...
  for( i = 0; i < MENU_ITEMS; i++ ) {
...
    u8g.drawStrP(1, i*h, menu_strings[i]);
  }
}

However the result is not what would one expect - on first draw instead of "First Line" there is "ép" and similar strange characters are on all lines. In addition they are changing on every draw.
I couldn't find any explanation which would describe how to use properly U8G_PROGMEM so it could be also my fault. Refering just to single value like u8g.drawStrP(1, i*h, string_0); works fine.
Tested on U8GLIB_SH1106_128X64 u8g(U8G_I2C_OPT_NONE);  and U8GLIB_ST7920_128X64_1X u8g(10); with same results.

MsiSoftwareEngineer, 2016-02-25:
Hmm i have never used the drawStrP...

For the bitmaps I use u8g_progmem, but for the text arrays i just use PROGMEM as below:

`const char  UpTempConfirmTextSV[] PROGMEM = "Bekr" "\xe4" "fta b" "\xf6" "rv" "\xe4" "rde " "\xf6" "ver";
const char* const UpTempConfirmText[] PROGMEM = {UpTempConfirmTextEN, ...`

Then I just copy the PROGMEM into RAM like this:

` strcpy_P(ErrorMessageLabel.title, (char*)pgm_read_word(&(SystemErrorText[CurrLanguage])));`

`	u8g.drawStr( leftAlignedLbl.pos.x, leftAlignedLbl.pos.y, leftAlignedLbl.title);`

I hope that helps. Can someone clarify if there is any benefit on using drawStrP like ig79 vs what i do?

Thanks,

igi79, 2016-02-26:
Thank you. My expectation from drawStrP was to save cycles and bytes in sram used for temporary variable. In Arduino u8g library is demo "F", which has different approach to PROGMEM drawStr.
The only example for drawStrP I found, is in m2tklib where it is utilized quite purposelessly - typecasting from regular string.

MsiSoftwareEngineer, 2016-02-26:
I think I tried at one point and it didn't work for me either igi79. You are not really saving neither cycles or bytes, since prior to sending the data over it must be loaded in the RAM to be used by the SPI/I2C bus. 

Great library olikraus.

olikraus, 2016-03-05:
I think the initial problem was that the pointer array also was placed in PROGMEM area. This requires additional code and is not a u8glib problem.