olikraus/u8glib #400: Arabic font

armantaherian, 2016-04-05:
Hey, first of all let me say that your library is amazing. :+1: 
So, I have a question about Arabic font, how we can use them? Can you help me please?
I mean this series:
[image]

Thanks.

olikraus, 2016-04-06:
Hi

Let me use "Hello World" as an example. Not sure whether google translate is right, but this should be this:
مرحبا بالعالم

Feeding this into https://r12a.github.io/apps/conversion/ or https://www.branah.com/unicode-converter will generate the following hex sequence:
0645 0631 062D 0628 0627 0020 0628 0627 0644 0639 0627 0644 0645

From these hex codes, you have to replace 06 with \x, which gives (after removing the spaces):
\x45\x31\x2D\x28\x27\x20\x28\x27\x44\x39\x27\x44\x45
This is actually what can be printed in the Arduino IDE:
u8g.print("\x45\x31\x2D\x28\x27\x20\x28\x27\x44\x39\x27\x44\x45");

I have not tested this, but I hope it will work.

If i ever get finished the next big release for u8glib, it will add support for UTF-8 and unicode glyphs, so that you can directly write:
u8g.print("مرحبا بالعالم");
Well.. it is a vision...

Oliver

steronydh, 2016-04-14:
Support for UTF-8 would be brilliant!

farzad321554, 2016-08-04:
Congratulations for your new release (U8g2) Oliver
We are looking forward for "Unifont_12_13" suppurt in this new version

olikraus, 2016-08-04:
Thanks, it is finally finished. Can you tell me which Unicode Ranges you need in the Arabic Font?
Total number of gyphs can exceed 256.

Thanks, Oliver

farzad321554, 2016-08-04:
Dear Oliver
Thanks for your rapid reply

I think for Persian language I need these Unicode 9.0 Ranges of the Arabic Font :
1) Arabic  0600-06FF
2) Arabic Presentation Forms-A  FB50-FBB1
3) Arabic Presentation Forms-B  FE70-FEFF 

Greetings from Iran
Farzad

olikraus, 2016-08-05:
Hi Farzad

Great, I have created issue 39 in u8g2 for this