olikraus/u8glib #403: Lcd initializes only when code is uploaded, not after power cycle

maverickoulos, 2016-04-14:
I am somewhat of a newbie to Arduino and programming and learning as I go.

I am using the current version of u8glib with an Arduino Nano clone (ch340) and a Nokia 84x48 LCD (PCD8544). I connected everything right, the display works great with the default Hello World example in both SW and HW SPI. I am using the appropiate constructor with all the pins, including reset.

The issue: If I upload the sketch from Arduino, the first run is ok. If i reset Arduino from the reset button it still works ok. If I unplug the USB (unpower) and plug it back (power on) the display doesn't load.
Tried it in both SW and HW SPI, it behaves the same.

As an added clue: upset that i cannot get it to work i forgot it plugged for 10-15 minutes or so after a power down - power up cycle and after such a long time the screen started showing something, writing vertical lines (something like 1by8 pixels) from top left towards top right, but at a very very slow rate.. I guess it's like a screen initialization or something.

As you know the SCK pin for SPI on Arduino is 13, the same with the integrated led. During normal operation the led blinks like crazy, but after the power cycle it blinks very very slowly.

I know it's not the hardware because I used the same nano with the same lcd running adafruit library and didn't had this issue. I don't like adafruit library and i'd rather stick with u8glib if i can tackle this issue.

Any idea what could be wrong and how could be solved?

A big thank you to all involved in developing and maintaining this great lib.

maverickoulos, 2016-04-15:
If I power the Arduino from an external trusted source (aka NOT the USB port..), everything runs fine.
For me this solves the problem, as this is the scope of the project, to finally use the Arduino and the lcd externally powered.
Still.. on the USB reconnect it didn't happen with the AF lib, so I think there is something with the code in u8glib also (maybe the data connection on the usb when reconnecting the Arduino interferes with the init of the display..)
Anyway, thank you for the great lib, this is now a minor issue for me

olikraus, 2016-04-30:
Excause me for not answering for such a long time. U8glib uses some low level code which is more faster than the normal SPI lib. Not sure what is used by the adafruit lib, but maybe this causes the problem.

closed 2016-04-30 by olikraus