olikraus/u8glib #408: Library is not very portable as it doesn't stick to the Arduino API

Muessigb, 2016-04-26:
Hello!

I am struggeling to use this library with my 86Duino which is Arduino IDE compatible but does obviously not have the registers of the standard ATmega series as it's an i386 based architecture. It does however support a special version of the Arduino IDE that comes with all standard API functions implemented. This isn't the first time I get this issue as it is probably the same with the ESP8266. I really strongly suggest adding some module that ONLY uses the standard Arduino API and not a single platform specific register call. Is this possible?

olikraus, 2016-04-29:
2.x version will be (much) more portable. See the u8g2 github project. In fact making u8glib more portable was a primary goal for u8g2. 
Bad news: I will not improve portability with u8glib 1.x. Please consider using u8g2 once we have a first release.
Can i close this ticket?

Muessigb, 2016-04-29:
That is a great announcement!
I will look into u8g2 as soon as possible.

I would love to contribute and help developing the new u8g2 if you don't
mind. I have worked with a lot of different microcontrollers over the last
time and have the hardware to debug most Atmel ARM and AVR uC's.
I can also test on various ESP8266 and 86Duino's.

You can close this issue if you want.

olikraus, 2016-04-29:
contribute... nur zu, jede Hilfe ist willkommen.

Once you have cloned this repo, you will find arduino examples in
u8g2/sys/arduino
You may need to update the makefiles, but in principle, the make files will upload directly to the target board.
What would be your prefered display for testing?

Muessigb, 2016-04-29:
Dann bin ich gerne dabei!

My preferred display for testing would be the ST7920 in SPI mode, but I can
also do the Nokia 5110 LCD and the SSD1306 OLED in SPI and I2C.

>
> contribute... nur zu, jede Hilfe ist willkommen.
>
> Once you have cloned this repo, you will find arduino examples in
> u8g2/sys/arduino
> You may need to update the makefiles, but in principle, the make files
will upload directly to the target board.

olikraus, 2017-01-15:
closing this, because u8g2 is there.

closed 2017-01-15 by olikraus