olikraus/u8glib #409: ATSAMD21 and SSD1306 don't work together

nine-thirtysix, 2016-05-02:
Hi,
I have a little trouble to get the Adafruit Feather M0 working with the OLED Displays. There is no compiling issue it just doesn't show anything (1.19.1). I tried all listed settings of the SSD1306 with I2C but no sign of life from the display.  
I got it working with the Adafruit libaries for SSD1306 Driver and 0x3C as I2C address, so the problem is somehow related to the u8glib.
I have no idea where i could start looking since I'm quite new to all this stuff.

Thanks in Advance,

nine-thirtysix, 2016-05-15:
I got it to work with the Version 2.0.1 since I could change the I2C pins easily.

olikraus, 2016-05-26:
yes, in fact i am testing u8g2 with the featherwing oled

nine-thirtysix, 2016-05-26:
I know you are still developing the u8g2. How similar to the U8g will it get? For now  I don't get the results I need. I could not use bitmaps and the fonts I knew from u8g.
Just quite a trouble because of the ATSAMD21 :-(

olikraus, 2016-05-26:
There is a feature compare list and update guide here:
https://github.com/olikraus/u8g2/blob/master/doc/u8g2.txt
If you name the object as u8g instead of u8g2, then it should be "mostly" compatible.
Bitmaps should also be the same.
There is also a font mapping table (see u8g2.h) included to be more compatible, so for many fonts you do not need to change the font name at all.

nine-thirtysix, 2016-05-27:
You are right I got it  to work mostly.
The font "u8g_font_freedoomr25n" does not work. I quite liked this one.

And one trouble I ran into is that if I want to read my weather sensor over I2C the display stops working, which it didn't with the u8g lib.

The specific part which breaks it is the following:

```
void readData()
{
  int i = 0;
  uint32_t data[8];
  Wire.beginTransmission(BME280_ADDRESS);
  Wire.write(0xF7);
  Wire.endTransmission();
  Wire.requestFrom(BME280_ADDRESS, 8);
  while (Wire.available()) {
    data[i] = Wire.read();
    i++;
  }
  pres_raw = (data[0] << 12) | (data[1] << 4) | (data[2] >> 4);
  temp_raw = (data[3] << 12) | (data[4] << 4) | (data[5] >> 4);
  hum_raw  = (data[6] << 8) | data[7];
} 

```
I haven't written this part and I don't know enough about all this to find a starting point for debugging.
I just know the display works without this sub programm.
Do you have a advice please?

nine-thirtysix, 2016-05-28:
That is strange...If i change the setup line to Hardware I2C it works....
That's perfect!

So only one wish left...could link the font u8g_font_freedoomr25n in u8g2 please?
Thank you!

olikraus, 2016-05-29:
https://github.com/olikraus/u8g2/issues/13

olikraus, 2016-05-29:
the freedoom fonts are now added to u8g2. Please check the new names for the font:

https://github.com/olikraus/u8g2/wiki/fntgrpu8g