olikraus/u8glib #412: SSD1306 I2C not working after power unplugged

Xylynx, 2016-05-21:
When I initially flash the code with the Arduino IDE the display works perfectly and as expected. However, as soon as I unplug power from the Arduino and then re-attach it the display stops working and the pin 13 LED flashes rapidly in double pulses for a few seconds then more slowly with singles pulses indefinitely. The display stays totally blank. If I reset the arduino with the onboard button or by removing power the exact same thing happens. Bizarrely, if I open the Serial monitor the display begins to function perfectly again.

Any got any ideas?

Thanks
 Nathan

jlwainwright, 2018-02-05:
I have the same issue. Ever got it resolved?

olikraus, 2018-02-05:
I rewrote U8glib completly. It is now U8g2 and available via the Arduino Library Manager.
My suggestion is to use U8g2. It fixes a lot of issues and problems with older U8glib.

U8g2 bug tracker is here: https://github.com/olikraus/u8g2/issues