olikraus/u8glib #413: STM32F103 - SSD1306 - u8glib - > 128 x 64 bit oled display - > Garbled Display on 2nd half of Scrren

disonance, 2016-05-24:
Hi, 

I've integrated u8glib into my code base and I'm able to display correctly on half the oled screen. However, I'm unable to access the 2nd half of the screen. IT always comes up garbled.

I'm using a STM32F103 board and I've tested the SSD1306 on an arduino. The hardware is okay.
Being unfamiliar with displays, I've only gotten as far as to check that the I2C comms are okay. It sends 8 pages of I2C data. I've not looked into if the data is correct or not. I've assumed that it must be, since I'm able to send correct strings and pixels up to the first half of the screen.

Its very puzzling. I've changed between I2C ports as well, since it was mentioned before that SWD could affect the I2C1 on the stm32F103. 

Has anyone seen this ? 

br
Diso.

OLED display.
-----------------------
*     ok     |   NOK  *  
*     ok     |   NOK  *
*     ok     |   NOK  *
-----------------------

disonance, 2016-05-24:
adding picture -> My previous attempts at showing the error in text was a diaster..
[image]

olikraus, 2016-05-24:
Very strange. Can you try the SH1106 constructor instead?
I recently had a similar issue with wire.h. It has a build in transfer buffer which is limited to 32 bytes. But i assume you example did not use wire.h (actually Arduino version of u8glib bypasses wire.h)

disonance, 2016-05-24:
> buffer which is limited to 32 bytes. But i assume your example did not use
> wire.h (actually Arduino version of u8glib bypasses wire.h... but u8glib
> also does not support STM by default)

Hi ,

I've been debugging this little further and I think it could be some
dependency on the way u8glib is written and how Keil cross compiles the
code.

If it were the transfer buffer, I should be able to fix it. I'll definitely
look into that, thanks.

I'm also getting into Memory access exception traps when I use this lib
function

u8g_draw_box ( u8g, 0, 0, 127, 40)
- > If I called the function with Width at 127, I get into a trap. If i had
called the function with Width below 100, the function goes thru and draws
a block
with a width of around 64 pixels and some random pixels after.

The processor I'm using has 64KB of flash, i'm using , with u8glib , no
more then 30KB...

pls let me know if u think of something..

br
Dennis

disonance, 2016-05-24:
as far as I can tell, its a there's no buffer for the I2C transfer.

The data goes via byte transfer on the I2C bus.

     while( arg_val > 0 )
        {
        I2C_SendData(I2C1, *ptr++);
        arg_val--;
        while(!I2C_CheckEvent(I2C1, I2C_EVENT_MASTER_BYTE_TRANSMITTED));
        }

I'm beginning to think that the I2C HWAL is okay, but the data being
written is wrong or corrupted somehow....

olikraus, 2016-05-25:
It is possible that there is a memory issue in the code. Unfortunately i do not have the equipment to do memory debugging on a microcontroller itself. However the linux version of u8glib runs without issues. 
Your I2C code is not part of the original library and there is also Keil support (i do not own this compiler). So at the moment i do not know how to support here :-(

Only thing what comes in to my mind: Maybe check the page buffer size, which is defined in the u8g_dev_xxx files along with the controller specific code.

One complete other option could be to move to version 2 of u8glib (named u8g2). It already has support for SSD1306 and is MUCH more portable. At least portability was one of the design goals for u8g2.

disonance, 2016-05-25:
Still debugging but it's definitely a memory allocation/corruption  issue.
I can see that the ptr buffer data is not per expected. Where this is
coming from, will take more time to figure out...

I'll definitely try out u8g2 lib, I'm just curious to why this u8g lib
isn't working per expected, so I'm gonna spend a little more time with the
old lib.

Quick question though, on the u8g lib , I wasn't able to find where to
increase the page buffer size. I'm using the ssd1306_128x64_i2c file. Am I
missing something ?

olikraus, 2016-05-25:
ssd1306_128x64_i2c.c???? Is it from this repo? I do not know this file.

disonance, 2016-05-25:
u8g_dev_ssd1306_128x64.c

Sorry, I wasn't exact on the file name before. It is from this repo.

olikraus, 2016-05-26:
yes, buffersize is defined there.

disonance, 2016-05-26:
found the bug, it's in my code implementation! U8glib works perfectly now !
Thanks

olikraus, 2016-05-26:
great.

charlespax, 2016-07-15:
@disonance I'm facing a similar issue. Can you provide more detail on what specifically was wrong with your code?

disonance, 2016-07-15:
Hi Charles,

Candan26, 2016-08-15:
Hello,
I have the same problem on my OLED screen(I2C,white,128x64) I am using stm32f4xx discovery with SSD1306 library. I am only able to use 106x64 pixels of the screen.When try to use 107x64 or higher pixel resolution my screen get stopped or  I only receive garbage pixels all over the screen. I  try to find the bug in code but I haven't get succeed  about this issue. I believe its a memory corruption in some how but as I mention about before I haven't solve the problem. 
Does any body face this kind of problem ?
Could any body help me ? 
I could send the code as well for any need. 
Thank in advance for your help
[image]

olikraus, 2016-08-15:
Which SSD1306 library do you mean?

Candan26, 2016-08-15:
I  am not quite sure about that but I could said that I follow up   this tutorial http://stm32f4-discovery.net/2015/05/library-61-ssd1306-oled-i2c-lcd-for-stm32f4xx/ . The library file is end of the tutorial.

olikraus, 2016-08-17:
Yes, but this is the issue tracker for u8glib, which seems to be a total different library.