olikraus/u8glib #414: OLED outputting junk values

peopledrivemecrazy, 2016-05-26:
I am trying to use the hello world example and the bottom portion of the screen seems to display junk values 
I have tried all these three
/U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NONE|U8G_I2C_OPT_DEV_0);	
//U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_DEV_0|U8G_I2C_OPT_NO_ACK|U8G_I2C_OPT_FAST); 
U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NO_ACK);	
[image]

olikraus, 2016-05-26:
Strange... Can you share the datasheet of the OLED?

peopledrivemecrazy, 2016-05-27:
[This is where I purchased from](http://www.aliexpress.com/item/Free-shipping-1Pcs-128X64-Blue-OLED-LCD-LED-Display-Module-For-Arduino-0-96-I2C-IIC/32658340632.html)
Alright, now it is strange, Now i woke up and turn it on the bottom portion is now clean. But I was using adafruit's for a while.

olikraus, 2016-05-29:
then... still not sure how to support here...

shridharnator, 2020-12-17:
I am having the same problem when I Input an Image.
`#include "U8glib.h" // include the universal graphcs library
 
U8GLIB_SSD1306_128X64 u8g(12, 11, 10, 9, 8);  // initialise a u8g display object
 
const uint8_t flux_xbm[] PROGMEM = {
   0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xBF, 0x56, 0x6D, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xEF, 0xFB, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x5F, 0x69, 0x92, 
  0x6E, 0xE7, 0xFF, 0xFF, 0xFF, 0xFF, 0x00, 0xC1, 0xFF, 0xFE, 0xFF, 0xFF, 
  0xFF, 0x3F, 0x80, 0x01, 0xF5, 0x7E, 0xFF, 0xFF, 0xFF, 0xFB, 0x00, 0xC2, 
  0x3F, 0x21, 0xF4, 0xFF, 0xFF, 0xFF, 0x36, 0x6C, 0xBE, 0x7F, 0xFC, 0xFF, 
  0xFF, 0xFF, 0x00, 0xC0, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x7F, 0x86, 0x61, 
  0xDE, 0x4F, 0xFE, 0xFF, 0xFF, 0xFB, 0x64, 0xE4, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x3F, 0x20, 0x04, 0xFC, 0x63, 0xFA, 0xFF, 0xFF, 0xFD, 0x24, 0xCD, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x95, 0xA9, 0x7F, 0x08, 0xFC, 0xFF, 
  0xFF, 0xF7, 0x27, 0xDA, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xBF, 0x00, 0x00, 
  0x04, 0x50, 0xFC, 0xFF, 0xFF, 0xFF, 0xD9, 0xC9, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x3F, 0x10, 0x01, 0x21, 0x90, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0x02, 0x84, 0x08, 0x02, 0xFC, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0x04, 0x00, 
  0x00, 0x28, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x1F, 0x00, 0x00, 0x00, 0x00, 0xF8, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0x00, 0x00, 0x00, 0x00, 0xFC, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0x00, 0x00, 
  0x00, 0x00, 0xF8, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x1F, 0x00, 0x00, 0x00, 0x00, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x00, 0x00, 0x00, 0x00, 0xF8, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0x00, 0x00, 
  0x00, 0x00, 0xF8, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x1F, 0x00, 0x00, 0x80, 0x07, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x00, 0x00, 0xC0, 0x05, 0xF8, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x00, 0x00, 
  0xB8, 0x0F, 0xF8, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x3F, 0x00, 0xE0, 0xFB, 0x1F, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x00, 0xF0, 0xFF, 0x0F, 0xF8, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x00, 0xBC, 
  0xFE, 0x1F, 0xF8, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x3F, 0x00, 0xDF, 0x49, 0x1F, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x00, 0x4B, 0xA8, 0x1F, 0xF8, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x40, 0x00, 
  0x40, 0x7A, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x1F, 0x80, 0x00, 0xA0, 0xFD, 0xF8, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0x40, 0x00, 0x00, 0x7A, 0xFC, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3F, 0x00, 0x00, 
  0xA2, 0xF5, 0xF9, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x1F, 0x70, 0x00, 0x00, 0xF9, 0xFD, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x30, 0x00, 0xA0, 0xFC, 0xFD, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x00, 0x00, 
  0x00, 0xE2, 0xF9, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x3F, 0x10, 0x00, 0x00, 0xD8, 0xFD, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x00, 0x00, 0xD2, 0xB7, 0xFB, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0xD0, 0x02, 
  0x60, 0x5F, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x1F, 0x08, 0x35, 0xF4, 0xF9, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x40, 0x08, 0xDC, 0xD4, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x20, 0x36, 
  0x16, 0x9F, 0xFD, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x1F, 0x42, 0xEC, 0xB7, 0xAB, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0xA0, 0x67, 0xAB, 0xBC, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1F, 0x40, 0x66, 
  0xB4, 0x6A, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x3F, 0x34, 0x49, 0x06, 0x95, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xDF, 0x0B, 0x20, 0x5C, 0x0A, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x1D, 0x10, 
  0x5C, 0xE4, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0x1E, 0x08, 0xB8, 0xD5, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x4A, 0x15, 0x64, 0x29, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x7F, 0x0E, 0x10, 
  0xAA, 0x91, 0xFD, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x7F, 0x9E, 0x00, 0xB4, 0x6C, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
[... 6 lines cut ...]
  0xAA, 0x7B, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xFF, 0x3F, 0x7F, 0xD8, 0xF9, 0xD5, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
  0xF7, 0xFF, 0xFF, 0xFF, 0xFF, 0xBF, 0xFF, 0x2A, 0xFA, 0xA8, 0xFD, 0xFF, 
  0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x7F, 0xFF, 0x4D, 
  0x6A, 0xB7, 0xF9, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
[... cut ...]
  0xBF, 0x9B, 0xEF, 0xFF, 0x5F, 0xAF, 0xAF, 0xAF, 0xAF, 0xAF, 0xBF, 0x43, 
  0x88, 0xFF, 0xFF, 0xFF, 0xFB, 0xFF, 0xFF, 0xFE, 0xFB, 0xFF, 0xFF, 0xFF, 
  0xFF, 0xFF, 0xEF, 0x41, 0xA8, 0xFF, 0x5D, 0x77, 0x9F, 0xBF, 0xE6, 0x77, 
  0xFF, 0xFB, 0xFB, 0xFB, 0xFB, 0xFB, 0xF7, 0xEF, 0xFE, 0xEF, 0xFF, 0xFF, 
  0xFF, 0xDB, 0xBF, 0xF7
};
 
void setup(void) {
}
 
void loop(void) {
  u8g.drawXBMP( 0, 0, 128, 64, flux_xbm); // send the xbm image to the display, specifying position and size
  while (1==1) {} // set the loop to only run once
}`

peopledrivemecrazy, 2020-12-17:
@olikraus  @shridharnator  Oh my, I forgot this issue is open! for me the I just had to change the jumper wires connecting to it, I believe it could be some grounding/connection issue.

closed 2020-12-28 by peopledrivemecrazy