olikraus/u8glib #416: Missing setContrast()

delinend, 2016-06-01:
Hi.

I missing the setContrast() funktion for my ssd1306 I2C display, in the NODEmcu build.
Is it possible that you can support  the setContrast() funktion in the u8glib for NODEmcu ?

Thanks and best regards.

olikraus, 2016-06-01:
it should be there. Which version do you use?

delinend, 2016-06-02:
I just see under "Unimplemented Functions" [http://nodemcu.readthedocs.io/en/dev/en/modules/u8g/#unimplemented-functions](url)

olikraus, 2016-06-02:
Maybe nodemcu uses an older version of u8glib. It is umplemented here in line 244:
https://github.com/olikraus/u8glib/blob/master/csrc/u8g_dev_ssd1306_128x64.c

devsaurus, 2016-06-02:
> Maybe nodemcu uses an older version of u8glib.

Correct, we integrate [release 1.18.1](https://github.com/olikraus/U8glib_Arduino/releases/tag/1.18.1) and I postponed binding `setContrast()` for the time being.
@delinend it will be included with the next update from upstream release 1.19.1.

olikraus, 2016-06-02:
thanks for the clarification.

delinend, 2016-06-03:
Thanks. I'm looking forward to use the u8glib build 1.19.1 on NodeMCU, when it's ready there.

delinend, 2016-07-09:
Do anyone knows, when the upstream release 1.19.1 is comming ?

olikraus, 2016-07-10:
Can't you select this from the library manager?

devsaurus, 2016-07-10:
@delinend I've started PR nodemcu/nodemcu-firmware#1385. Feel free to test drive the update and feed back any issues over there.

delinend, 2016-07-27:
@devsaurus Thanks. How do we get it over to nodemcu/nodemcu-firmware (dev), so that I can test it, via http://nodemcu-build.com/ ?

devsaurus, 2016-07-27:
@delinend The update won't be available from nodemcu-build before it's merged into the official firmware repo. Thus I've prepared images with a snapshot of the PR branch at
https://github.com/devsaurus/nodemcu-firmware/releases/tag/u8g_1.19.1
It contains the default modules plus the upgraded u8g.

Feel free to try them and feed back your results over at https://github.com/nodemcu/nodemcu-firmware/pull/1385.

devsaurus, 2016-08-02:
@delinend the update was merged to nodemcu upstream in the meantime.

delinend, 2016-08-08:
Thanks @devsaurus . I have tested it now, and have a small question. 
I'm using a small 0.96" oled (ssd1306_128x64) from Ebay via I2C and tryed setContrast(255) vs setContrast(0). There is a differency (it's working), but it is small. When using setContrast(0), it still bright my room up in the night. Is this an issue in the oled 0.96" display, or mayby in the code ? I want to Dim it more, but maybe the display can't do that?

devsaurus, 2016-08-08:
> When using setContrast(0), it still bright my room up in the night. Is this an issue in the oled 0.96" display, or mayby in the code ?

The code itself is pretty simple, thus I don't expect issues there. But you could measure the supply current of the display and compare it with the I<sub>SEG</sub> spec of the datasheet for different contrast settings. You'd need a quite precise current meter that's able to detect a delta of 50-80&nbsp;µA.

olikraus, 2016-08-09:
Have a look at this file:
https://github.com/olikraus/u8glib/blob/master/csrc/u8g_dev_ssd1306_128x64.c
It contains the init sequence for the OLED, which looks like this:

static const uint8_t u8g_dev_ssd1306_128x64_adafruit3_init_seq[] PROGMEM = {
```
  U8G_ESC_CS(0),             /* disable chip */
  U8G_ESC_ADR(0),           /* instruction mode */
  U8G_ESC_RST(1),           /* do reset low pulse with (1*16)+2 milliseconds */
  U8G_ESC_CS(1),             /* enable chip */

  0x0ae,				/* display off, sleep mode */
  0x0d5, 0x080,		/* clock divide ratio (0x00=1) and oscillator frequency (0x8) */
  0x0a8, 0x03f,		/* */

  0x0d3, 0x000,		/*  */

  0x040,				/* start line */
  
  0x08d, 0x014,		/* [2] charge pump setting (p62): 0x014 enable, 0x010 disable */

  0x020, 0x002,		/* 2012-05-27: page addressing mode */
  0x0a1,				/* segment remap a0/a1*/
  0x0c8,				/* c0: scan dir normal, c8: reverse */
  0x0da, 0x012,		/* com pin HW config, sequential com pin config (bit 4), disable left/right remap (bit 5) */
  0x081, 0x0cf,		/* [2] set contrast control */
  0x0d9, 0x0f1,		/* [2] pre-charge period 0x022/f1*/
  0x0db, 0x040,		/* vcomh deselect level */
```
You could play around with some parameters. For example you could disable the charge pump
( use 0x08d, 0x010) or you can redurce the vcomh deselet level, e.g. try 0x0db, 0x030 or 0x0db,0x020