olikraus/u8glib #417: fix t6963 code

olikraus, 2016-06-29:
I'm using: u8glib_arduino_v1.18.1.zip

With "U8GLIB_SSD1306_128X64" and it works great!

I noticed a lot of warnings in your code (see attachment).
I've changed the "-w" to "-Wall" in "platform.txt". I also added "-Wextra"
I like the extra warnings when developing my code.
Most can be ignored but I thought you might like to know about a couple of them.

These two:
U8glib/utility/u8g_com_arduino_t6963.c:255:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if ( x < millis() )
            ^
U8glib/utility/u8g_com_arduino_t6963.c:274:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if ( x < millis() )
            ^
Might create problems when an Arduino is running for a long time. I think it is > 49 days.

Changing "long x;" to "unsigned long x;" won't solve the problem.

I think you should use:

http://playground.arduino.cc/Code/TimingRollover

Maybe the code should be:

  unsigned long x = millis();
...
  if ((millis() - x) >= U8G_STATUS_TIMEOUT)
    return 0;

As:

https://www.arduino.cc/en/Tutorial/BlinkWithoutDelay

suggests.

Just thought you might like to know.

Thanks again for a great library.

Joe Szedula
P.S. I ran into the SPI conflict when using an RTC (DS3234).
Simply moving the display to it's own SPI buss works as the notes I found on-line suggested.
I tried "u8g.setHardwareBackup(u8g_backup_avr_spi);" but that didn't seem to work. I guess I didn't understand how to use it correctly.

snapedumbledore, 2018-02-23:
@olikraus hello, i also tried"u8g.setHardwareBackup(u8g_backup_avr_spi);"
i can't understand how to use it.
IDE1.8.5  12864ST7920  SD
can you help me, thanks

olikraus, 2018-02-23:
I do not support u8glib any more. Better use u8g2.