olikraus/u8glib #418: why my oled display like this (video)

luxin88, 2016-07-01:
video is at here https://youtu.be/n84YrjfmKZU  

my code write like this
````c++
#include "U8glib.h"
#include <Wire.h>
#include <Adafruit_BMP085.h>

U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NONE | U8G_I2C_OPT_DEV_0); // I2C / TWI
Adafruit_BMP085 bmp;

void draw(void) {
  u8g.setFont(u8g_font_helvR12);  //delay(500);
  char alt_buf[20];
  float alt = bmp.readAltitude(101100);
  char altStr[6];
  dtostrf(alt, 4, 2, altStr);
  sprintf(alt_buf, "ALT: %sm", altStr);

  u8g.setFont(u8g_font_helvR12);  //delay(500);
  int32_t pre = bmp.readPressure();
  float hpa = pre / 100.0;
  char preStr[6];
  dtostrf(hpa, 4, 2, preStr);
  char pre_buf[20];
  sprintf(pre_buf, "%shpa", preStr);

  u8g.setFont(u8g_font_helvR12);  //delay(500);
  float tmp = bmp.readTemperature();
  char tmp_buf[20];
  char tmp_str[6];
  dtostrf(tmp, 4, 2, tmp_str);
  sprintf(tmp_buf, "TMP: %s*C", tmp_str);

  u8g.setPrintPos(0, 20);
  u8g.print(alt_buf);
  u8g.setPrintPos(0, 40);
  u8g.print(pre_buf);
  u8g.setPrintPos(0, 60);
  u8g.print(tmp_buf);
}

void initScreen(void) {
  if ( u8g.getMode() == U8G_MODE_R3G3B2 ) {
    u8g.setColorIndex(255);     // white
  }
  else if ( u8g.getMode() == U8G_MODE_GRAY2BIT ) {
    u8g.setColorIndex(3);         // max intensity
  }
  else if ( u8g.getMode() == U8G_MODE_BW ) {
    u8g.setColorIndex(1);         // pixel on
  }
  else if ( u8g.getMode() == U8G_MODE_HICOLOR ) {
    u8g.setHiColorByRGB(255, 255, 255);
  }
}

void initBmp(void) {
  if (!bmp.begin()) {
    Serial.println("Could not find a valid BMP085 sensor, check wiring!");
    while (1) {}
  }
}

void setup() {
  Serial.begin(9600);
  initScreen();
  delay(50);
  initBmp();

}

void loop() {
  u8g.firstPage();
  do {
    draw();
  } while (u8g.nextPage());
  delay(1000);
}
````

olikraus, 2016-07-01:
because you update the sensors within the firstpage/nextpage loop
split the draw procedure into two parts: sensor reading & u8g drawing, then execute the sensor reading before the firstPage call.

BTW: U8glib V2 (u8g2) offers a mode without picture loop.

Oliver

luxin88, 2016-07-02:
thanks, i'll try it out

luxin88, 2016-07-02:
Thanks again, this problem has troubled me for three days, now resolved

closed 2016-07-02 by luxin88