olikraus/u8glib #419: BDF converter for OSX

christophepersoz, 2016-07-13:
Hello,

Is that possible to have the binary of BDF2U8G compiled for OS X ?
Thanks !

olikraus, 2016-07-13:
For U8glib or U8g2?
Font format is different :-(

For U8glib the source is just a single file, so i guess you can compile it by yourself:
https://github.com/olikraus/u8glib/tree/master/tools/font/bdf2u8g

For U8g2 the source is little bit more complicated, but you just have to compile and link all .c files:
https://github.com/olikraus/u8g2/tree/master/tools/font/bdfconv

I also do not have any OS X here, so i unfortunately can not do this for you :-(

maciekish, 2022-09-01:
This is old but i was able to compile bdfconv on macOS in case anyone needs it. It's very simple however, make sure you have Xcode and the command line tools installed, then its just a matter of **`cd`**:ing into the folder and **`make`**.

This is the Apple Silicon version binary [bdfconv.zip](https://github.com/olikraus/u8glib/files/9469418/bdfconv.zip)

Schokobecher, 2024-04-05:
Even older now, but I made a Universal binary for macOS x86_64/arm64

[bdfconv_macos_universal.zip](https://github.com/olikraus/u8glib/files/14880773/bdfconv_macos_universal.zip)