olikraus/u8glib #420: ssd1305

olikraus, 2016-07-24:
Für den SSD1305 würde ich folgende modifikationen im gegensatz zum 1306 einbauen (falls du die config aufnehmen willst):
In der init_seq3
0x021, 0x001, 0x081,  //column address set to 1 - 129
(da bin ich mir noch nicht sicher wenn ich es auskommentiere geht es auch)

0x0db, 0x034,            /* vcomh deselect level */ //0x000 = .43xVcc, 0x034 = .77xVcc, 0x03c = .83xVcc  (this affects display voltage and brightness (and burn in)) 00, 34(default), 3C possible (Datasheet Solomon systech)

    case U8G_DEV_MSG_CONTRAST:
      u8g_SetChipSelect(u8g, dev, 1);
      u8g_SetAddress(u8g, dev, 0);          /* instruction mode */
      u8g_WriteByte(u8g, dev, 0x081);
      u8g_WriteByte(u8g, dev, (*(uint8_t *)arg));
      u8g_SetChipSelect(u8g, dev, 0);
      break;

static const uint8_t u8g_dev_ssd1305_128x64_data_start[] PROGMEM = {
  U8G_ESC_ADR(0),       /* instruction mode */
  U8G_ESC_CS(1),        /* enable chip */
  0x010,                /* set upper 4 bit of the col adr. to 0 */
  0x002,                /* set lower 4 bit of the col adr. to 4  */ //********************************* IMPORTANT!!!  CHANGED from 0x000 to 0x004 for SSD1305 to fix offset!!! smithy change to 2 2016
  U8G_ESC_END           /* end of sequence */
};

E-Mail von Ingmar

jensnielsen, 2016-09-21:
Hi

For the record, I'm using a Densitron DD-12864WO-3A 128x64 display with SSD1305 and I get a 2 pixel offset with this setup. Quite possibly something going on that I don't understand but I have to have _0x010, 0x000_ in data_start to not get pixel offset. I've been toying with the 0x21 command to try to remap so there would be a generic solution for 128 pixel SSD1305 displays but that doesn't seem to make a difference for me...

DD-12864xx-3A are about to be obsolete and replaced with one with SSD1309 so there's no big deal I suppose but I just wanted to fill in with my experiences

olikraus, 2016-09-24:
thanks for reporting.
The ssd1305 is indeed not supported by u8glib. Hope i can add support to u8g2