olikraus/u8glib #421: Different Problem LCD Full Graphic 12864

dexac, 2016-08-18:
Many more repraper southeast asia buying this LCD, but till now, im can't found to solved this problem

I check at https://www.youtube.com/watch?v=4CD8ERaylmY someone modify it, but less so we understand.

U8GLIB_ST7920_128X64_1X u8g(23, 17, 16); // SPI Com: SCK = en = 23, MOSI = rw = 17, CS = di = 16

Please help us,

olikraus, 2016-08-18:
The pin number (here: 23, 17, 16) have to fit to your configuration. There is nothing what i can do on U8glib side. You have to figure out the correct pin numbers by yourself. I do NOT know your hardware.

dexac, 2016-08-18:
like this https://www.google.com/url?sa=i&rct=j&q=&esrc=s&source=images&cd=&cad=rja&uact=8&ved=0ahUKEwjror2DwcrOAhXDJJQKHZOZCfoQjBwIBA&url=http%3A%2F%2Fs1.electrodragon.com%2Fwp-content%2Fuploads%2F2014%2F06%2FRamps-Full-Graphic-Smart-Controller-Reprap-6.jpg&psig=AFQjCNFDfFMUO3MuDeh3rWCWWkRw1AIfrA&ust=1471594000943997, but different with western hardware, may from china

with define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER can show bitmap and home indicator, but rotary encoder won't work...

olikraus, 2016-08-18:
Again: You have to update the pin assignment in the REPRAP software. There is nothing to in U8glib (you are currently in the U8glib issue tracker).
I suggest to open a issue ticket for the REPRAP software (if a issue tracker exists for that software).

Basically you have to do what the person did in the video: Figure out the pins for your system.

olikraus, 2016-08-18:
Update: It looks like that RAPRAP does not have an issue tracker, but instead a forum. 
Your question is not related to u8glib, so please ask in the RAPRAP forum for support.