olikraus/u8glib #423: Library linking error

TechGuy254, 2016-08-28:
Hey I have included the library in the project folder yet i still get the following errors. What could be wrong? Am sure its something petty but I cant seem to solve it. I am using Atmega 16

main.o: In function `draw':
D:\project\Coding Progress\oled display/main.c:24: undefined reference to `u8g_font_7x14'
D:\project\Coding Progress\oled display/main.c:24: undefined reference to `u8g_font_7x14'
D:\project\Coding Progress\oled display/main.c:24: undefined reference to `u8g_SetFont'
D:\project\Coding Progress\oled display/main.c:25: undefined reference to `u8g_DrawStr'
D:\project\Coding Progress\oled display/main.c:26: undefined reference to `u8g_DrawStr'
D:\project\Coding Progress\oled display/main.c:27: undefined reference to `u8g_DrawLine'
D:\project\Coding Progress\oled display/main.c:28: undefined reference to `u8g_DrawLine'
main.o: In function `u8g_setup':
D:\project\Coding Progress\oled display/main.c:10: undefined reference to `u8g_Pin'
D:\project\Coding Progress\oled display/main.c:10: undefined reference to `u8g_SetPinInput'
D:\project\Coding Progress\oled display/main.c:11: undefined reference to `u8g_Pin'
D:\project\Coding Progress\oled display/main.c:11: undefined reference to `u8g_SetPinLevel'
D:\project\Coding Progress\oled display/main.c:12: undefined reference to `u8g_Pin'
D:\project\Coding Progress\oled display/main.c:12: undefined reference to `u8g_SetPinOutput'
D:\project\Coding Progress\oled display/main.c:14: undefined reference to `u8g_Pin'
D:\project\Coding Progress\oled display/main.c:14: undefined reference to `u8g_SetPinInput'
D:\project\Coding Progress\oled display/main.c:15: undefined reference to `u8g_Pin'
D:\project\Coding Progress\oled display/main.c:15: undefined reference to `u8g_SetPinLevel'
D:\project\Coding Progress\oled display/main.c:16: undefined reference to `u8g_Pin'
D:\project\Coding Progress\oled display/main.c:16: undefined reference to `u8g_SetPinOutput'
D:\project\Coding Progress\oled display/main.c:18: undefined reference to `u8g_dev_ssd1306_128x64_i2c'
D:\project\Coding Progress\oled display/main.c:18: undefined reference to `u8g_dev_ssd1306_128x64_i2c'
D:\project\Coding Progress\oled display/main.c:18: undefined reference to `u8g_InitI2C'
main.o: In function `main':
D:\project\Coding Progress\oled display/main.c:39: undefined reference to `u8g_FirstPage'
D:\project\Coding Progress\oled display/main.c:47: undefined reference to `u8g_NextPage'
make.exe: *** [main.elf] Error 1

olikraus, 2016-08-29:
you need to link the u8g c code with your project

TechGuy254, 2016-08-30:
Hi olikraus!! Thanks for your fast response. You mean u8g.c? I cant find it in the u8glib folder. am using ssid1306 128x64 and I want to use i2c with avr.

olikraus, 2016-08-31:
i mean all the files of the u8g project.
I guess your setup is wrong. Where did you download u8g?

TechGuy254, 2016-08-31:
https://bintray.com/olikraus/u8glib/AVR
I have included all files from the src folder to my project folder and it is still not working
[image]
[image]

olikraus, 2016-09-03:
This is probably not enough. You need to name the files or the directory and probably also the include dir in your project config. Please consult the documentation of you development envionment.
This is NOT a u8g problem.