olikraus/u8glib #426: Add horizontal flip

goocastro, 2016-09-11:
We are working on a project where we need to flip the entire screen.
When working with shapes and primitives, it's ok, since it's just a matter of math.

But texts are not ok.
It would be nice if the lib could support something like `setHFlip` and the screen flipped horizontally, something like `setRot90`, `setRot180`... but flipping the entire screen.

olikraus, 2016-09-12:
You mean, the text should appear mirrored? So to say, readable only with a mirror?

goocastro, 2016-09-12:
Exactly.
Some wearables, like the one we are prototyping, might require that.

I managed to achieve the result I wanted playing with adafruit's library. 
C isn't really my main language, so the code I wrote is really simple, but did the trick for now:

```
void Adafruit_SSD1306::flip(void){
  for (int i=0; i<8; i++) {
    reverse(buffer, SSD1306_LCDWIDTH*i, SSD1306_LCDWIDTH*i+(SSD1306_LCDWIDTH-1));
  }
}
```
I simply iterate through the buffer reversing the array.

goocastro, 2016-09-13:
Actually, I got the starting idea from here:

http://nerdclub-uk.blogspot.com.br/2016/03/adding-flip-function-to-adafruits.html?m=1

But it only rotates the screen, feature already in u8g.

I`d like to contribute to the repo, but as I said, it's not my main language. It's just a feature request. 😀

olikraus, 2016-09-13:
you could try to change the value 0x0c8 to 0x0c0 in line 130 of u8g_dev_ssd1306_128x64.c (assumming you have this display)

goocastro, 2016-09-13:
Yap, that's my display.
I'll try that tonight.

olikraus, 2016-09-19:
any success?

goocastro, 2016-09-20:
Perfect!
It worked. Thank you very much.

johnny-82, 2018-07-06:
Hi olikraus!
sorry for my English.
I need to flip horizontally my old "nokia 5210 lcd". I am using U8GLIB_PCD8544.
Can you help me change the code? do I have to edit the file "u8g_dev_pcd8544_84x48.c"?

olikraus, 2018-07-06:
what about using setRot180?

olikraus, 2018-07-06:
ok, I see. What about using u8g2 (which is u8glib version 2). It includes a mirror option should solve the problem.

johnny-82, 2018-07-06:
thanks a lot! It work with "U8G2_PCD8544_84X48_F_4W_SW_SPI u8g2(U8G2_MIRROR, /* clock=..."

olikraus, 2018-07-06:
HW SPI might be better

SteveC00k, 2018-11-24:
I have a similar issue. The text is mirrored.
I used this display setting: 

> u8g_dev_sbn1661_122x32.c

I altered the lines 54 and 59 to 0x0a and it did not mirror. See gfx below.

Here is what it looks like with the settings before altering.
[image]

Here is what it looks like after altering.
[image]

Back of LCD.
[image]

[image]

SteveC00k, 2018-11-24:
Here is the datasheet if anyone needs to see it. https://drive.google.com/open?id=1gbZHPdsXN4bUGbe_5cXVszRSPaIAwO8v

SteveC00k, 2018-11-25:
Started to use the ug82 lib.
I managed to get it working except half the text is chopped off on one of the examples.

SteveC00k, 2018-11-25:
I have it working now. Found out how to get it going using this profile in the examples.
U8G2_SED1520_122X32_1 u8g2(U8G2_MIRROR, 8, 9, 10, 11, 4, 5, 6, 7, /*dc=*/ A0, /*e1=*/ A3, /*e2=*/ A2, /* reset=*/  U8X8_PIN_NONE);   // Set R/W to low!

closed 2020-01-19 by gcstr

gdattn, 2025-02-13:
I had such a problem with flipped screen while trying to connect 3d printer mainboard MKS-Robin-E3D to COG screen ST7567 with Marlin firmware. Problem solved thanks to this thread. Changes in \Marlin\src\lcd\dogm\u8g_dev_uc1701_mini12864_HAL.cpp in line 71:
change
#define UC1701_OUT_MODE(N)       ((N) ? 0xC8 : 0xC0)
to
#define UC1701_OUT_MODE(N)       ((N) ? 0xC0 : 0xC8)