olikraus/u8glib #430: arduino --install-library "U8glib" failed

thinkyhead, 2016-09-20:
I'm seeing this in Travis-CI when we try to use the Arduino built-in install method. For some reason the command is failing, citing a bad CRC.
```
Verifying archive integrity...
java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:77)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:97)
	at processing.app.Base.<init>(Base.java:415)
	at processing.app.Base.guardedMain(Base.java:223)
	at processing.app.Base.main(Base.java:136)
The command "arduino --install-library "U8glib"" failed and exited with 255 during .
```
Is it possible the version of u8glib stored at arduino.cc has been corrupted?

thinkyhead, 2016-09-20:
I've notified Arduino also. arduino/Arduino#3206

olikraus, 2016-09-24:
I am not working much on u8glib (instead there is version 2, u8g2). But maybe i should create another release to ensure, that the library is valid.

thinkyhead, 2016-09-24:
I altered our `.travis.yml` to download directly from Github and it works well. Might even be faster, who knows?

facchinm, 2016-10-03:
Should be solved now by rebuilding the whole library index. Thanks for reporting!

closed 2016-10-05 by thinkyhead