olikraus/u8glib #432: No support for Chip select 3 (CSEL3) for KS0108 192X64 screen

RndMnkIII, 2016-10-03:
Hi. I've using screens with KS0108 controller which uses 3 chip select lines for a 192x64 resolution. I've tried to modify source code of u8glib with no success. Can you help me to guide how to fix source code to include support for three chip select lines?

RndMnkIII, 2016-10-03:
The chip select logic is active on low:
                    CSEL1      CSEL2         CSEL3
Chip1 sel        0                 1                  1
Chip2 sel        1                  0                 1
Chip3 sel        1                   1                0

RndMnkIII, 2016-10-03:
I've using a Teensy 3.2 MCU and without modifications the screen it's update 2/3 of full size, replicating the 64-127 screen colums in the 128-191 columns but with glitches.

olikraus, 2016-10-04:
Difficult. It would require major changes to u8glib. Moreover i have almost stop development for u8glib, because i wanted to concentrate more on the the successor u8g2 (u8glib v2.x).

Jet code for KS0108 has not been ported to u8g2:
 * There is a great library for KS0108 displays: https://bitbucket.org/bperrybap/openglcd/wiki/Home
 * Request for this display type was very less
 * Especially your type of KS0108 is very rare

RndMnkIII, 2016-10-04:
Yeah, openglcd I've tested but have serious problems with screen clipping, for rendering a pixel smooth text flyer side scroller. Your library render (with 16 bit support enabled for screen width) the text with screen clipping ok. The openglcd support the CSEL3 chip select. I'll try modify u8glib another time... :-(

olikraus, 2016-10-04:
hmm... yes... proper clipping was one of the design goals for u8g libs...
Maybe you want an issue for u8g2... i still could add support for KS0108 there...

RndMnkIII, 2016-10-04:
What is the difference between these versions and how is selected the version to use?
u8g_com_arduino_fast_parallel.c, 
u8g_com_arduino_no_en_parallel.c,
u8g_com_arduino_parallel.c

RndMnkIII, 2016-10-04:
How can add a new  device as  u8g_dev_ks0108_192x64.c to include new modifications? Need to define/declare in another file/class?

RndMnkIII, 2016-10-04:
If for example I known that in teensy 3.x I can use at least 8 bits of the same port, how I take advantage of this?

RndMnkIII, 2016-10-04:
For write 8 bits of data at same time

olikraus, 2016-10-05:
ok, two different question.
parallel 8 bit write: Basically you can write a new low level com function.
See for example u8g_com_arduino_fast_parallel_fn in 
https://github.com/olikraus/u8glib/blob/master/csrc/u8g_com_arduino_fast_parallel.c

u8g_dev_ks0108_192x64.c:
Basically you can copy this from u8g_dev_ks0108_128x64.c. Then add one more block for the third graphics controller in the init part (third controller needs to be enabled) and the transfer part (transfer data to the third controller). In the transfer part you need to choose one more value for the thired chip select. Maybe use
u8g_SetChipSelect(u8g, dev, 3);
However this new value has to be considered in your new COM function.
See line 182ff in u8g_com_arduino_fast_parallel.c
In fact in your case you have to consider three chip select lines.
One option is to just hardcode this in u8g_com_arduino_fast_parallel.c. I mean you could just use digitalWrite here. Another option is to go through all the code and extend everything to support three chip select lines. But this is a little bit difficult ...

RndMnkIII, 2016-10-06:
Thank you, I'll try yo hardcode changes first, I've understand that with ks0108 parallel interface the u8g_com_arduino_fast_parallel.c is used.

RndMnkIII, 2016-10-06:
Tested with an teensy 3.x overclocked up to 144Mhz your library works ok and the speed up is more or less proportional.

RndMnkIII, 2016-10-16:
I fixed source code of u8glib for 3 chip select lines suport. Mainly I modified the U8GLIB adding an additional constructor to pass uint8_t cs3 paramater and making new functions u8g_Init8Bit2, init8Bit to include cs3. Modified u8g_com_arduino_parallel_fn to include new case in U8G_COM_MSG_CHIP_SELECT, and added new device u8g_dev_ks0108_192x64 and new class U8GLIB_KS0108_192. All modifications in:
[https://github.com/RndMnkIII/U8glib ](https://github.com/RndMnkIII/U8glib )

Would lack to add check code for test when te CS3 pin is defined or is not used, to modify u8g_com_arduino_parallel.c or the function used to write to the the pins, enable the CS, etc. Also lack better integration with other devices.

closed 2016-10-16 by RndMnkIII