olikraus/u8glib #433: Drawing string with more than 255px width will cause it to overlap

endlessmind, 2016-10-05:
I'm trying to display strings sent over bluetooth and scroll them if they where longer than the 128px my display has.
While using the 10x20 font, I noticed that some string would overlap.
So part of the string would be draw over the beginning of the string.

When investigating this, I found that string longer than 25 char (25 = 250px) work start to overlap.
So, if the string was 30 char, the last 5 char would be written on top of the first 5. (Imagen a clear piece of tape with text on it wrapped around a bottle. If the tape is to long, you would end up with more than 1 layer of tape at some places)

Is there any solution for this? Would it help to switch over till u8g2?

olikraus, 2016-10-06:
Yes, same would happen with u8g2.
For both libs you have to enable 16 bit mode.
See the beginning of u8g.h for this.
In 16 bit mode i guess you would have about 6553 chars. 
However: Expect some delay for drawing very long strings. It might be a clever idea to select the visible substring first instead of fully relying on the u8g clipping procedure.

endlessmind, 2016-10-06:
Okey, thanks!
I had an idea for scrolling substring, so even if the library didn't have 16 bit mode I would manage.
My string will most likely never be longer than 40 chars, so can't be much of a delay while drawing.

olikraus, 2016-10-06:
so you are saying, it will not work?
it may depend also on your calling code.
maybe you can post the code for the scrolling part.
very important: you have to use the internal type u8g_uint_t for all kind of variables holding x/y coordinates.

endlessmind, 2016-10-06:
This is the code I'm using.
`
u8g_uint_t title_scroll_x = 0;
//In function
setBigFont();
  u8g_uint_t strLen = u8g.getStrWidth(title);
  u8g.drawStr(title_scroll_x, 2, title);
  
  u8g.setFont(u8g_font_6x13r);
  u8g.drawStr(0, 30, artist);
  if(strLen > 128) {
    if (title_scroll_x > (128 - strLen)) {
        title_scroll_x--;
    } else if (title_scroll_x == (128 - strLen)) {
      title_scroll_x = 128;
    }
  } else {
    title_scroll_x = 0;
    if (title_scroll_x == 0)
      menu_redraw_required = 0;
  }
  char StringLenChar[13];
  sugmig(StringLenChar, strLen);
  u8g.drawStr(0, 45, StringLenChar);`

Before the title_scroll_x  variable was just a uint8_t, but when I swiched to u8g_uint_t that scroll stopped working. Now it's just static position. Also, getStrWidth returns the length - 255 if string is longer than 255 even if I've uncommented `#define U8G_16BIT 1`

endlessmind, 2016-10-06:
Solved it!
Did 3 things:
Change title_scroll_x  to long
Changed strLen to long
and replaced `u8g.getStrWidth(title)` with `strlen(title) * 10` (because I'm using the 10x20 font.

Quick question, I live in Sweden so I've got the 3 nordic letters: åäö. But I have no idea how to encode them so that u8glib can display them.

olikraus, 2016-10-07:
Hi
The tearing is probably caused by the page height of u8glib. You could test with the 2x or 4x version of the same display device.
Your letters should be included in the font,
please crosscheck here: https://github.com/olikraus/u8glib/wiki/fontgroupx11

nordic letters:
U8glib does not support unicode. This was introduced with u8g2.
Instead you have to provide the 8-bit hex code in the string:

å = "\xe5"

You could also derive the hex code from the font overview picture (see link above).
With u8g2 it would be indeed simpler: You could just write "from a to å" instead of "from a to \xe5"

Please note that not all fonts include all glyphs. So please check whether  your letter is really included.

endlessmind, 2016-10-07:
Thanks! 
Not sure what you mean with:

>  You could test with the 2x or 4x version of the same display device.

I'm using `U8GLIB_SH1106_128X64 u8g(U8G_I2C_OPT_DEV_0|U8G_I2C_OPT_FAST);` right now.
What would I use for `the 2x or 4x version`?

olikraus, 2016-10-07:
Yes, just try U8GLIB_SH1106_128X64_2X instead of U8GLIB_SH1106_128X64

endlessmind, 2016-10-07:
Thanks! That fixed the tearing!

Thanks for a great library!

endlessmind, 2016-10-07:
Sorry for all the questions, but when I replace "å" with "\xe5" the display prints "xe5"

olikraus, 2016-10-07:
Strange. See bottom of page 59 of this document: http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1124.pdf

hexadecimal-escape-sequence:
\x hexadecimal-digit hexadecimal-escape-sequence  hexadecimal-digit

And later in section 6.4.4.4, clause 6:
The hexadecimal digits that follow the backslash and the letter
x
in a hexadecimal escape sequence  are  taken  to  be  part  of  the  construction  of  a  single  character  for  an  integer character  constant  or  of  a  single  wide  character  for  a  wide  character  constant.   The numerical  value  of  the  hexadecimal  integer  so  formed  specifies  the  value  of  the  desired character or wide character.

So i guess something else is wrong...

endlessmind, 2016-10-07:
I was using "u8g_font_10x20r", but when I swiched to "u8g_font_10x20" I still get "xe5".
I've also tried uppercase the hex-code, but no change.
"this is what the arduino received "P\xe5" and the display shows "Pxe5"

olikraus, 2016-10-07:
Can you paste the command here?
I mean the string.

endlessmind, 2016-10-07:
Okey. So, The arduino receive a json string like this:
`{"T":1,"N":"01 - Flickorna P\\xe5 Tv2","A":"<unknown>"}`

After the json har parsed the string, I get:
`const char* title = 01 - Flickorna P\xe5 Tv2:` (but of course the title is a global variable)

The output on the display is now `01 - Flickorna P\xe5 Tv2`

Before I started typing this, I remembered that json escapes the string, so I added another backslash.
But the output is now as above where before it was `01 - Flickorna Pxe5 Tv2`

olikraus, 2016-10-07:
oh, you do not have a hard coded string? \xe5 is a feature of the c-compiler.
So you receive some json, right? As far as i remember json is UTF8 encoded, so you need a UTF8 decoder which converts to 0xe5. You could also send for example $ and convert this to å just before printing, assuming that $ is not used.
So i guess your problem is not u8glib specific but is more how to transfer nordic chars though some communication channel.
BTW: u8g2 support UTF8 out of the box...

endlessmind, 2016-10-07:
Yes, I'm using UART between the bluetooth and the Arduino. If i print the input from the bluetooth (Serial1.read()) to the usb-serial (Serial.print()) the nordic letters is there correctly. So i assume that there is no problem with nordic letters in UART. So, should I convert for example å to 0xe5 before sending or do I need to convert it on the Arduino?

olikraus, 2016-10-08:
you could just send the 0xe5 via UART.

endlessmind, 2016-10-08:
Hm, that doesn't work either :/

Sending: `{"T":1,"N":"01 - Flickorna P0xe5 Tv2","A":"<unknown>"}`
Display showing: `01 - Flickorna P0xe5 Tv2`

Also tried adding a backslash before it: \0xe5
But It just printed it like that.

endlessmind, 2016-12-11:
Solved it.
But converting the json data to ascii I was able to get the nordic letter (and other) working.
Found some UTF8 to ASCII convertion code on arduino.cc