olikraus/u8glib #435: U8G font format is abstract.

alidaf, 2016-10-10:
Where is the font format defined so that I can adapt my own font generator to create the correct format?

olikraus, 2016-10-10:
there is a bdf to u8g converter which includes some description:
https://github.com/olikraus/u8glib/blob/master/tools/font/bdf2u8g/bdf2u8g.c

Instead of writing your own generator, perhaps better create a bdf file, which is then processed by bdf2u8g
bdf is a standard ascii format for fonts.

Also note that there is u8g2 (u8glib v2.x) which has a new font format and also supports unicode.

alidaf, 2016-10-10:
I saw that file but it's a bit hard to decipher without some images to describe what is what. I'm dyslexic, which doesn't help. I started trying to write a graphics library for the SSD1322 using an rPi some time ago and stumbled a bit at the fonts. I came across the Adafruit GFX library and wrote a spreadsheet that I could create the glyphs in, and then export the data. I coded up a routine to pack the data into the Adafruit format. I'd hoped to use the spreadsheet to do the same. Plus, I also like to know how things work! The main problem I have is finding something that I can use to design the font, export to bdf, and then convert to U8G, all on Linux. I've found something called Fony that runs in Wine so I'll give that a go but I'll keep trying to decode the format for my own edification.

closed 2016-10-10 by alidaf

olikraus, 2016-10-10:
well, yes, coding all the font handling things had been the most time consuming part.
Writing a good documentation for the font format would probably be as time consuming as coding. I finally tried to spend more time on end user documentation,which i guess is more important.