olikraus/u8glib #437: It isn't clear how 4bpp images are supposed to be drawn.

alidaf, 2016-10-12:
The Rook example only gives an example of a monochrome graphic. I have a graphic array that is essentially an array of colour indices but using drawBitmapP decodes it as if it were 1bpp values.

olikraus, 2016-10-12:
Yes. correct.

alidaf, 2016-10-14:
In one of your screenshots you have a logo that looks like 4bpp. How did you do that? I've gone back to my driver, which worked on the rPi but I just can't get it to work either!

olikraus, 2016-10-14:
can you link the picture?

alidaf, 2016-10-14:
https://github.com/olikraus/u8glib/wiki/otherpic/nhd_oled180.jpg
It looks like 3 different greys there so I guess 2bpp rather than 4 but I remembered that it wasn't monochrome, or perhaps you've used a trick to give the illusion!

olikraus, 2016-10-14:
This is not a bitmap, but just constructed with graphic primitives. Text and lines are drawn with different gray levels one after the other.

alidaf, 2016-10-14:
OK, I'll try to sort my own code out then. Thanks.

closed 2016-10-14 by alidaf