olikraus/u8glib #439: Impossible to get this working in Atmel Studio

MCUdude, 2016-10-20:
Hi!
I've been using the u8glib library for my Arduino projects for a long time, and it's allways been easy to set up. I'm slowly migrating to Atmel Studio 7 instead, and wanted to create a project where I'm using u8glib. I've loaded up the "HelloWorld" example, and added u8glib to the Solution explorer, so that Atmel studio is able to find the library. I've also added the correct #include path.

Here's the error message
```
cannot find -lu8g.h  
recipe for target 'OLED2.elf' failed  
Id returned 1 exit status
```

What is this, and how can it be solved?

here's the code (main.c)
``` c
#include <avr/io.h>
#include <avr/interrupt.h>
#include <avr/sfr_defs.h>
#include "u8glib/src/u8g.h"

u8g_t u8g;

void sysInit(void)
{
	/* select minimal prescaler (max system speed) */
	CLKPR = 0x80;
	CLKPR = 0x00;
}

void lcdSetup()
{
	u8g_InitSPI(&u8g, &u8g_dev_ssd1306_128x64_sw_spi, PN(1, 2) /* PB2 */, PN(1, 3) /* PB3 */, PN(1, 4) /* PB4 */, PN(1, 5) /* PB5 */, PN(1, 6) /* PB6 */);
}

void draw(void)
{
	u8g_SetFont(&u8g, u8g_font_6x10);
	u8g_DrawStr(&u8g, 0, 15, "Hello World!");
}

int main(void)
{	
	// Power the LCD using PB0 and PB1
	DDRB |= _BV(PB0) | _BV(PB1);
	PORTB |= _BV(PB0);
	PORTB &= ~_BV(PB1);
	
	// Init LCD
	sysInit();
	lcdSetup();
	
    /* Replace with your application code */
    while (1) 
    {
		u8g_FirstPage(&u8g);
		do
		{
			draw();
		} while ( u8g_NextPage(&u8g) );
		u8g_Delay(100);
    }
}
```

olikraus, 2016-10-24:
looks like a atmel studio issue.
-lu8g.h is for sure wrong. Maybe -lu8g works
I once had created a tutorial for this, but since shutdown of googlecode this is broken and i never found time to fix this. also the pics are there somewhere but not linked correctly
https://github.com/olikraus/m2tklib/wiki/as6

MCUdude, 2016-10-24:
> -lu8g.h is for sure wrong. Maybe -lu8g works

where (in Atmel Studio) can I change this?

olikraus, 2016-10-24:
i never used atmel studio 7, but i guess there is a dialog box on linker settings and options. somewhere there.  
However: -lu8g means, that atmel studio has to create a u8g-library (-lu8g just means to include u8g library). The extension usually is u8g.lib or u8g.a (not sure about this). So u8g.lib or u8g.a has to be constructed somehow. As long as this lib does not exist, still -lu8g will fail with a linker error.

ApoorvSingh0009, 2020-05-18:
[image]
I am also facing the same issue with the linker. Can you please help me how to fix this?
@olikraus

olikraus, 2020-05-22:
this does not look like an u8g problem.