olikraus/u8glib #440: CAN bus and OLED - display not updating

n1md4, 2016-10-22:
Hi,

  I'm using several modules to create a CAN bus network.  At the moment I have 2 nodes, one creating an input and the other with the OLED attached displaying the output.

```
/*

   OLED
   ===
   1....GND: (arduino pin GND)
   2....VCC: (arduino pin 5V)
   3....D0: CLK Clock (arduino pin 13)
   4....D1: MOSI data (arduino pin 11)
   5....RST: Reset (arduino pin 5)
   6....DC: data / command (arduino pin 6)
   7....CS: Chip select signal (arduino pin 7)

   CAN controller
   ===
   1....Int NC
   2....SCK (arduino pin 13)
   3....MOSI (arduino pin 11)
   4....MISO (arduino pin 12)
   5....CS (arduino pin 7)
   6....GND (arduino pin GND)
   7....Vcc (arduino pin 5V)

*/

#include "U8glib.h"            // U8glib library for the OLED
#include <SPI.h>
#include "mcp_can.h"

const int SPI_CAN_CS_PIN = 10;
MCP_CAN CAN(SPI_CAN_CS_PIN);                                    // Set CS pin

U8GLIB_SH1106_128X64 u8g(3, 4, 5);  // CS = 3, A0 = 4, RST = 5

unsigned int messageData[2];
unsigned int displayData[2];

void draw(void) {
  u8g.setFont(u8g_font_unifont);                   // select font

  u8g.setPrintPos(0,24);
  u8g.print(messageData[0]);
  Serial.println(messageData[0]);

  u8g.setPrintPos(0,48);
  u8g.print(messageData[1]);
  Serial.println(messageData[1]);*/
}

void displayVariables(void) {
  displayData[0] = messageData[0];
  displayData[1] = messageData[1];
}

void readCAN(void) {
  
  unsigned char messageLength = 0;
  unsigned char messageBuffer[2];
  
  CAN.readMsgBuf(&messageLength, messageBuffer);
  messageData[0] = CAN.getCanId();
  messageData[1] = messageBuffer[1];
}

void drawOLED(void) {
  displayVariables();
  u8g.firstPage();  
  do {
    draw();
  } while( u8g.nextPage() );
}

void setup() {
  
  Serial.begin(9600);
  CAN.begin(CAN_500KBPS);          // init can bus : baudrate = 500k

}

void loop() {

  readCAN();
  drawOLED();
  delay(2000);
}
```

If I change the test text to display it will not update the display until I comment-out this line `CAN.readMsgBuf(&messageLength, messageBuffer);`, including, for example, if I remove the array data and just add a "hello world" drawstr command.  It would appear the display data is buffered and does not change, with my code.

I know the data to display should not change before being rendered, and hence the displayVariables() function.

olikraus, 2016-10-24:
The old library u8glib does not really work together with SPI.h.
Switching to u8g2 would be a much better option here.
https://github.com/olikraus/U8g2_Arduino

n1md4, 2016-10-24:
Perfect.  Along with the porting guide, this now works perfectly.