olikraus/u8glib #444: Grayshade Bitmap draw

christophepersoz, 2016-11-03:
Hi Oliver,

Actually drawXBMP() draw a XBM bitmap in black and white (0/1) values. Do you think it can be possible to draw bitmap in gray shade on displays that support more color values ?

Thanks

olikraus, 2016-11-03:
I have almost stopped development for u8glib and I will concentrate on u8g2.
The only drawback is, I have stopped graylevel support completly to simplify the software architecture. So... i do not support this any more in u8glib and i will not support graylevel in u8g2 (although some minimal graylevel support might be thinkable).

christophepersoz, 2016-11-03:
Too bad, I can understand that means heavier code. I will have a look on your code if there is a way to display a bitmap in grayscale or not.
In my proper opinion, grayscale increase readability, and allow to focus on some parts of the display. So for my current app I will stay on U8Glib thought ;)

Thanks for your quick reply!

closed 2016-11-03 by christophepersoz

olikraus, 2016-11-03:
You are the first person ever who had asked for graylevel support. I had assumed, that this is of almost no use. For my own projects I also never needed graylevel support.

christophepersoz, 2016-11-03:
Funny. No prob ;)