olikraus/u8glib #446: "inverse" color for monochrome OLED

fanfanlatulipe26, 2016-11-21:
For a monochrome OLED display, setColorIndex allows 0 (black) and 1 (white).
Is it possible to have the "inverse" color, the new value of the pixel depending on the previous value (xor operation) ?
This feature available in an other well known Arduino library is usefull for drawing for examples ticmarks or scale grids on a graphic.

olikraus, 2016-11-21:
Maybe for u8g2, but not for u8glib

fanfanlatulipe26, 2016-11-21:
Thank you Oliver. I just made some try with u8g2 but the library seems a bit bigger and it is a problem for me. I just need to work a bit more ;-)

olikraus, 2016-11-22:
I try to concentrate on u8g2 only at the moment...

fanfanlatulipe26, 2016-11-22:
Yes I understand. Congratulations for your great work.
As a lighth weight but nevertheless powerfull library, u8glib is very useful for my development. 
No doubt that u8g2 will also be a success.
Many thanks.

----- Original Message ----- 
  From: olikraus 
  To: olikraus/u8glib 
  Cc: fanfanlatulipe26 ; Author 
  Sent: Tuesday, November 22, 2016 9:37 AM
  Subject: Re: [olikraus/u8glib] "inverse" color for monochrome OLED (#446)

  I try to concentrate on u8g2 only at the moment...

remy68, 2018-03-29:
"better late than never..." Thank You for this awesome work. I have found a mean to invert text display with library u8glib:
...
u8g.drawBox(x, y, w,h);
u8g.setColorIndex(0);
u8g.drawStr(z ,v,  "Some text");
u8g.setColorIndex(1);
...
of course drawStr must be centered on the box.
Works on U8GLIB_SSD1306_128X64

olikraus, 2018-03-29:
u8g2 already supports xor mode for drawing...