olikraus/u8glib #449: SSD1306 128X64 working with SW but not HW?

formantjim, 2016-12-03:
I have used u8glib for a number of projects with both the UNO and Mega with great success. I need to use the edge connector at the edge of the Mega pcb so wanted to move the display from the software serial pins 13,11,10,9 and 8 to hopefully the HW SPI connection ie pins 50 through 53 on the Mega bur cannot get it to work.
Looking at the selection available in the u8g2 library these are the only choices available.
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 12, /* dc=*/ 4, /* reset=*/ 6);	// Arduboy (Production, Kickstarter Edition)
//U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

As you can see they now only use 3 pins do we simply ignore MOSI and CLK now?

olikraus, 2016-12-03:
hw SPI pins are hardwired. U8g2/u8glib will use the same pins as SPI.h.

formantjim, 2016-12-04:
Thanks for the reply but it still does not work. Maybe a bit more explanation of what I have got and how I should be wiring it up let alone selecting the correct HW driver line of code to use.
I currently have both a UNO and Mega together with the SSD1309 diplay that when using your example sketch Graphic Test in u8g2 **work perfectly** when the uncommented line U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); past is used.
But when trying to use either the only selections in the sketch //U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 12, /* dc=*/ 4, /* reset=*/ 6);	// Arduboy (Production, Kickstarter Edition)
//U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); it does not.
I maybe missing something here completely but if I use the four pins assigned to the SPI 50,MISO 51,MOSI 52, CLK and 53, CS I just get a blank display. But given only 3 pins are available in the uncommented lines above how does one connect this display?
I have looked at the code in SPI.h and I could not see mention of the pins to be able to change so can you please guide me please?

Cheers JIM

olikraus, 2016-12-04:
Basically, you have to provide the pin numbers to the constructor. I mean, just uncommenting the constructor is not enough, you have to change the pin values (i think i have described this in the tutorial on the wiki for u8g2)

Your pins are this: 
SPI 50,MISO 51,MOSI 52, CLK and 53, CS
Then there are two points:
1. How can U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 13, / data=/ 11, / cs=/ 10, / dc=/ 9, / reset=/ 8); work?
The numbers do not match at all.
2. MISO is NOT used by u8g2 and u8glib. Instead your display requires the dc and reset input. How are these pins wired?

> I have looked at the code in SPI.h and I could not see mention of the pins to be able to change so can you please guide me please?
Of course not, because these pins are hardwired to the mikrocontroller. 
Pins are mentioned here:
https://www.arduino.cc/en/Reference/SPI

Your problem discription is still incomplete. Please describe: Which board is wired in which way and how does the working constructor look like.

One more little thing: Do not use the slave select pin at all. Better use any other pin for chip select. This is not a u8g2 problem but a hardware issue with the atmega family.

formantjim, 2016-12-04:
Thank you so much for taking the time to reply, in answer to your questions see below.
_How can U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 13, / data=/ 11, / cs=/ 10, / dc=/ 9, / reset=/ 8); work?
The numbers do not match at all._
This indeed does work **as programmed and connected** to my display in fact it is running your Display Test as we speak on my bench! The sketch with this line uncommented and the pins as they are.
So if I understand this correctly if I use your sketch Display Test which line do I uncomment?
My display has the following pins SCL,(13) SDA,(11) RES(10 but not used),DC(9) and CS(8) and has a link on the PCB to select I2C or SPI and SPI is connected. Currently what I use and have used for four projects with displays. So according the the SPI Arduino information 51 =MOSI goes to SDA on my PCB, CLK goes to SCL which of the two lines in the sketch Display Test do I use?  Obviously with the pin numbers being changed to suit pins available on the edge connector? Really appreciate you taking the time to assist here.

olikraus, 2016-12-04:
> My display has the following pins SCL,(13) SDA,(11) RES(10 but not used),DC(9) and CS(8)

So the values in ( ) denote the Arduino pin number to which the display is connected, right?
In this case you can use the following constructor:

```
U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 13, / data=/ 11, / cs=/ 8, / dc=/ 9, / reset=/ 10); 
```
For the Uno (and only for the Uno) you can use this:
```
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, / cs=/ 8, / dc=/ 9, / reset=/ 10); 
```
This is possible, because on the Uno pin 13 is the clock of the internal HW SPI and pin 11 is the MOSI line of the internal HW SPI.

formantjim, 2016-12-04:
Thanks Olikraus so as I understand it I'm not able to reassign the pins in the line
U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 13, / data=/ 11, / cs=/ 8, / dc=/ 9, / reset=/ 10); 
to this to work with the Mega?
U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ **52**, / data=/ **51**, / cs=/ **47**, / dc=/ **48**, / reset=/ **49**);

olikraus, 2016-12-04:
U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 52, / data=/ 51, / cs=/ 47, / dc=/ 48, / reset=/ 49);

Well, looks better. Does this work? Of course your display needs to be wired in the way you have provided the pins.

formantjim, 2016-12-04:
Perfect it works!!!!!

Thank you so much could not see the wood for the trees!!!!!!

olikraus, 2016-12-04:
Now what about HW SPI? Your display seems to be wired correctly. So you just need to change the constructor name and remove clock and data pins.

formantjim, 2016-12-06:
That was going to be my next question. Unfortunately I'm away on business but will try tomorrow.
One comment i will make is I'm using this display with the GP3906 the update times have increased in performance. I'm hoping the latency will be reduced further with the HW version. Will report later.

formantjim, 2016-12-06:
Well report is here, the new pins and HW settings work perfectly!
I think the new U8g2 is fantastic thank you so much. You keep the world of DIYers very happy indeed!