olikraus/u8glib #452: How to add accents to charset?

mrbbp, 2017-01-11:
Hello, i'm french and i use extended ascii char a,e,i,o,u with accents and some ligature œ,æ or ç.
How to add char to font with more chars?

olikraus, 2017-01-12:
This is one of the reasons, why i stopped working on u8glib and created u8g2 instead: U8g2 support UTF-8
1. Migrate to u8g2.
2. Use a font, ending in "f" (full char set)
3. Use the accent/ligatur in your string: u8g2.print("œ,æ or ç");

gitterman, 2018-07-08:
I tried your example with the u8g2_font_tom_thumb_4x6_mf and the u8g2_font_5x7_mf but both did not print the degree symbol nor the square exponent symbol.
`      n5110.setFont(u8g2_font_tom_thumb_4x6_mf);    
	n5110.setCursor( 20,30 ) ;
	n5110.print( "I²C @ 27°" ) ;`

olikraus, 2018-07-08:
Did you enable UTF-8 for the print command?

https://github.com/olikraus/u8g2/wiki/u8g2reference#enableutf8print

gitterman, 2018-07-08:
Thanks for the very quick answer. 
I found that the problem came from the vsvim editor I use with visual studio.
If I use the arduino IDE the UTF characters are coded correctly and show up on my display.
Obviously there is some setting wrong which sets the character encoding in vsvim.
Thanks for your great library.

olikraus, 2018-07-08:
Yes, the IDE must generate correct UTF-8 code.

gitterman, 2018-07-08:
aside from its xx_mf name the u8g2_font_tom_thumb_4x6_mf does not produce a degree or a square symbol though. I was looking for a small font as the nokia 5110 only has 84x48 dots.
I think I can use a mixture of fonts though for my purpose.

olikraus, 2018-07-08:
the degree sign and "²" are  both included:

[image]