olikraus/u8glib #454: UC1701 Mini12864 not working on DUE

GarrethX, 2017-02-26:
I'm trying to set up display UC1701 MINI12864 for 3d printer on DUE board, and I have issues.

So far from my tests:
1. Hello world sketch works for SW SPI ( U8GLIB_MINI12864 u8g(76,75,45,46,42); )
2. Hello world is not working for HW SPI (U8GLIB_MINI12864 u8g(45,46,42); )
3. Hello world is working ok at u8glib2 (U8X8_UC1701_MINI12864_4W_HW_SPI u8x8(45,46,42); )

I know that u8glib is somewhat deprecated, however firmware I'm basing on (repetier) would be extremely hard for me to port to u8glib2. 

Maybe there is some easy fix for the u8glib to make it running on DUE for this display?

olikraus, 2017-02-26:
One of the reasons for the development of U8g2 had been support for HW SPI on Due and other advanced Boards. 

I think Marlin people are already working on that port to U8g2.

GarrethX, 2017-02-26:
Yeah, I opened an issue on the repetier firmware https://github.com/repetier/Repetier-Firmware/issues/640 , it would be great if you can give a little bit of contribution there about the memory usage and speeds. 

However maybe for now you could give me some hints where to look inside the code to make my uc1701 mini12864 display with the old version of the library? I remember I tested one OLED HW SPI screen on DUE with u8glib and it had no problems, also being was very fast.

olikraus, 2017-02-26:
I have checked the code and I wonder why HW SPI is not supported.

GarrethX, 2017-02-26:
Well, the best I achieved in the HW SPI was a black screen (all pixels ON), but I will try to play with it this week, hopefully with some success. Maybe SW SPI will be good enough for repetier, but I really hoped for this HW implementation to be working. So far I've got monkeys in my head about it:)

MsiSoftwareEngineer, 2017-02-26:
The reset pin has given us trouble before with a different display. Try driving it high.

GarrethX, 2017-02-27:
Actually I was mistaken - RS pin is not connected to the display, so low or high doesn't matter, unless you mean trying to find a place on the display board to put additional pullup on a reset pin (if its even there). 

There is something inside u8glib implementation of HW SPI for this display wrong. As I mentioned, on the u8glib2 it works perfectly, also SW SPI works good.