olikraus/u8glib #455: Arduino Micro and SSD1322 issues

BiatuAutMiahn, 2017-03-11:
Hello, I am attempting to get [this](http://www.ebay.com/itm/131619441678?_trksid=p2060353.m1438.l2649&ssPageName=STRK%3AMEBIDX%3AIT) to work, but i keep having issues.
I uncommented "U8GLIB_NHD31OLED_GR" in the example, and set all of the pins accordingly, and nothing happens, I have tried using the actual SPI pins on the micro with no success either.

olikraus, 2017-03-11:
maybe you can do the test with u8g2

https://github.com/olikraus/u8g2/wiki/gallery#25-jun-2016-st7920-128x64-lcd

BiatuAutMiahn, 2017-03-11:
is MSOI not needed?

BiatuAutMiahn, 2017-03-11:
I get an error for the Hello World Example:
```
Sketch uses 12210 bytes (42%) of program storage space. Maximum is 28672 bytes.
Global variables use 2688 bytes (105%) of dynamic memory, leaving -128 bytes for local variables. Maximum is 2560 bytes.
```

BiatuAutMiahn, 2017-03-11:
Ok, this can be closed, it seems there is a faulty connection on my display...U8G2 works beautifully!

olikraus, 2017-03-11:
ok...

closed 2017-03-11 by olikraus