olikraus/u8glib #456: SSD1322 SetContrast Not working.

BiatuAutMiahn, 2017-03-13:
Hello, I am attempting to set the contrast/brightness on [this](http://www.ebay.com/itm/131619441678?_trksid=p2060353.m1438.l2649&ssPageName=STRK%3AMEBIDX%3AIT), but it does not seem to respond to the command.

Thank you

olikraus, 2017-03-13:
Development has moved forward to u8g2. Can you check with u8g2?

BiatuAutMiahn, 2017-03-13:
I would use u8g2, however it does not support grayscale on my module

olikraus, 2017-03-13:
You are right, u8g2 does not support grayscale any more. 

You could change the code by youself.

https://github.com/olikraus/u8glib/blob/master/csrc/u8g_dev_ssd1322_nhd31oled_gr.c

lines 252ff

not sure what is not working there...

BiatuAutMiahn, 2017-03-13:
hmm, i'll have to compare the changes from this version. U can go ahead and close this, thank you!