olikraus/u8glib #457: OLED 1.3" based on SH1106  - displays dots after initialisation

bgolab, 2017-03-22:
Hi,
Just connected the oled display to STM32 throuh I2C. Unfortunately it display random dots after initialisation.
-tested several drivers ssd1306, sh1106 - no success
-I have no problem with oled 0.96" based on ssd1306

So it seems there is an issue with initialisation this 1.3" display. 
For arduino there are additional options to use with constructor. How to enable I2C option for ARM ?
I mean for example this one U8G_I2C_OPT_NO_ACK

Any other suggestion?

bgolab, 2017-03-22:
The display I am talking about:
http://www.ebay.com/itm/192000063292

bgolab, 2017-03-22:
The code I used for init:
u8g_InitComFn(&u8g, &u8g_dev_sh1106_128x64_i2c, u8g_com_hw_i2c_fn);
And directly after executing this line the dots appear.

olikraus, 2017-03-23:
Can you test with u8g2?
I don't maintain u8glib any more.

bgolab, 2017-03-23:
Hi,
Thank you for your comment.

I switched to the nucleo-F411 board and I do not see the issue anymore. Probably it was hardware issue as the previously tested board has some passive elements soldered and they may disturb the I2C communication.

The only problem I see now u8g_dev_sh1106_128x64_i2c is used) is displayed dots at the last column of the oled display.

Bogdan

bgolab, 2017-03-23:
I use very simple code:
 u8g_SetFont(&u8g,u8g_font_fur11);	//set current font  
      u8g_DrawStr(&u8g, 2, 12, "Hello!");		//write string - you set coordinates and string  
      u8g_DrawBox(&u8g, 30, 30, 35, 35);		//draw some box 
Everything is displayed correctly but the last column has random dots enabled.

bgolab, 2017-03-23:
The dots in the right last column are displayed after the init. Sometimes some random dots are displayed also in the next column as well (to the left from the last column).

bgolab, 2017-03-23:
Tested also the u8g_dev_ssd1306_128x64_i2c driver - no change.

bgolab, 2017-03-23:
It seems that column 0 and 1 is not displayed (wrote simple program displaying pixel and x coordinate).
The last columns are out of control (the content is random).

It is strange because theoretically the sh1106 is supported by the u8g lib.

bgolab, 2017-03-23:
Is u8g2 better if it comes to handling the sh1106?

bgolab, 2017-03-23:
The last column I can draw a pixel is 126. The 127th cannot be displayed.

bgolab, 2017-03-23:
I missed something:
-when I select the sh1106 the picture is somewhat messy( garbled with pixel noise)
-when I select ssd1306 the picture CANNOT show column 0 and 1 and the last 2 columns have random pixels.

So it seems that the driver sh1106 does not work for this display but the ssd1306 works ok but has problem with some columns.
I wonder what is the chip inside;)

bgolab, 2017-03-23:
I modified the ssd1306 driver as follow:
static const uint8_t u8g_dev_ssd1306_128x64_data_start[] PROGMEM = {
  U8G_ESC_ADR(0),           /* instruction mode */
  U8G_ESC_CS(1),             /* enable chip */
  0x010,		/* set upper 4 bit of the col adr to 0 */
  // 0x000,		/* set lower 4 bit of the col adr to 0  */
  0x002,		/* set lower 4 bit of the col adr to 2   - MODIFIED!!!! */
  U8G_ESC_END                /* end of sequence */
};

and now I can see the column 0 and 1 BUT still cannot see the column 127 (it contains random pixels).

bgolab, 2017-03-23:
FINALLY:
static const uint8_t u8g_dev_ssd1306_128x64_data_start[] PROGMEM = {
  U8G_ESC_ADR(0),           /* instruction mode */
  U8G_ESC_CS(1),             /* enable chip */
   0x010,		/* set upper 4 bit of the col adr to 0 */
  // 0x000,		/* set lower 4 bit of the col adr to 0  */
  0x003,		/* set lower 4 bit of the col adr to 3   - MODIFIED!!!! */
  U8G_ESC_END                /* end of sequence */
};

With the settings above:
-no random pixel is seen
-column 0 and 1 are visible
-column 127 is not accessible - I cannot draw a pixel in this column
The modified ssd1306 driver is used for an oled 1.3" with sh1106 chip (at least this info is on ebay)

olikraus, 2017-03-24:
Not sure how to support here from my side.
U8glib is not supported any more that much... also it looks a little like you have a very unusual

bgolab, 2017-03-24:
u8g2 is to consider. Not sure if there is any manual how to setup it with the ARM (basic data structure to use, etc).

bgolab, 2017-03-24:
I spent  couple of hours today studying the data sheet and crafting the driver myself. I succeded (no random pixel, stable display). The working init looks as follow:
void SH1106Init(void)
{
// I2C Initialisation	
HAL_Delay(100);	

SH1106SendCommand(SH1106_DISPLAYOFF);               //0xae, Display OFF
	
SH1106SendCommand(0x02);														//0x02,Low Column shifted because the RAM has 132 columns
SH1106SendCommand(0x10);														//0x10,//High Column

SH1106SendCommand(SH1106_SETSTARTLINE);       			// 0x40, line #0, start line

SH1106SendCommand(0xB0);														// set page address

SH1106SendCommand(SH1106_SETCONTRAST);              // 0x81, set contrast
SH1106SendCommand(0x80);

SH1106SendCommand(0xa1); 														//0xA1,//remap 

SH1106SendCommand(0xa4);								            // 0xA4
SH1106SendCommand(SH1106_NORMALDISPLAY);            // 0xA6, normal display 

SH1106SendCommand(SH1106_SETMULTIPLEX);             // 0xA8, multiplex ratio
SH1106SendCommand(0x3F);                            // 63, 1/64 duty 

SH1106SendCommand(0xAD);          									// DC-DC ON
SH1106SendCommand(0x8B);    

SH1106SendCommand(0x33);														// VPP=9.0V

SH1106SendCommand(0xc8); 														// scan direction

SH1106SendCommand(SH1106_SETDISPLAYOFFSET);         // 0xD3, display offset
SH1106SendCommand(0x0);                             // no offset

SH1106SendCommand(SH1106_SETDISPLAYCLOCKDIV);       // 0xD5, Display clock divide
SH1106SendCommand(0x80);                            // the suggested ratio 0x80	

SH1106SendCommand(SH1106_SETPRECHARGE);             // 0xd9, precharge period
SH1106SendCommand(0x22);

SH1106SendCommand(SH1106_SETCOMPINS);               // 0xDA, com pins 
SH1106SendCommand(0x12);

SH1106SendCommand(SH1106_SETVCOMDETECT);            // 0xDB, VCOM deselect
SH1106SendCommand(0x40);

// SH1106SendCommand(0x20);
// SH1106SendCommand(0x02);

HAL_Delay(100);	

SH1106SendCommand(SH1106_DISPLAYON);								//0xAF, display ON  

HAL_Delay(100);	
}

I am not sure if the delay is required. It seem s that there is only paging mode. So when I wanted to clear or refresh the display I need to walk through all pages:
void OLED_Clear(void)  
{  
	int i,n;		    
	for(i=0;i<8;i++)  
	{  
		SH1106SendCommand(0xb0+i);
		SH1106SendCommand(0x02);
		SH1106SendCommand(0x10); 
		for(n=0;n<128;n++) SH1106SendData(0); 
	}
}

void OLED_Refresh(void)  
{  
	int i,n;		    
	for(i=0;i<8;i++)  
	{  
		SH1106SendCommand(0xb0+i);
		SH1106SendCommand(0x02);
		SH1106SendCommand(0x10); 
		for(n=0;n<128;n++) SH1106SendData(_SH1106buffer[i*128+n]); 
	}
}

JavierPascual, 2017-05-04:
@bgolab where are you setting this code?

philpugh51, 2017-06-29:
I also have this problem with generic 1.3" 128*64 display.  I have working code using 0.96" 128X36 display that will 'work' on this display (in FRAME mode and only in top half using the construct
U8G2_SSD1306_128X32_UNIVISION_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);  //128*32

Using the construct below I get a one or two pixel column missing from the LHS of display and a random 1or 2 columns on RHS display.
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE); //128x64

I am using the U8G2 library and the graphics demo will work with these annoying 'features'  Is there a simple fix do you think?

Phil

WilkoV, 2018-06-11:
It looks like, that I have the same problem with an cheap 1.3" oled display and an esp8266 (NodeMCU 1.0). After I tried some options, it looks like U8G2_SH1106_128X64_NONAME_1_HW_I2C works in most cases. Most just because I haven just tested some text based samples.

WilkoV, 2018-06-12:
Had problems with UTF8 fonts (like in the weather example) and my new favorite for a ESP8266 and a cheap SH1106 1.3" oled display is:

U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);

olikraus, 2018-06-12:
I do not work in u8glib any more.
Please recreate a ticket here: https://github.com/olikraus/u8g2/issues

Also provide some details about your problems.
What is your code? What do you see? What do you expect?